MagicMirrorOrg/MagicMirror #2468: replace request

khassel, 2021-02-27:
The `request` package is deprecated, see https://www.npmjs.com/package/request

So we should replace it with e.g. `node-fetch`.

It is used in 
- calendarfetcher.js
- newsfeedfetcher.js
- and several e2e tests

Can submit a PR, but want to discuss this before if this is wanted and if `node-fetch` is the right replacement.

fewieden, 2021-03-01:
I also started moving my modules to `node-fetch` a while ago

MichMich, 2021-03-01:
That sounds like a good idea!

khassel, 2021-03-01:
thanks, will start working on this.

khassel, 2021-03-05:
after looking into `package-lock.json` I realised that some dependencies are also still using `request`. These are
- `node-ical`
- `spectron` (only dev dependency)
- `jsdom` (only dev dependency)

So I can send my PR but instead of removing dependeny count we would get 4 more. Any suggestions?

rejas, 2021-03-05:
I would still go for it. Having it replaced here is helpful. Waiting for the other libraries to remove it would take too much time. 

Maybe one could look if one can help the libraries (especially node-ical since it is the only non.dev dependency) to replace request?

MichMich, 2021-03-13:
Closing this since it's merged. Let me know if I need to reopen.

PR #2474 (2021-03-13): Node fetch
closes #2468

This PR replaces the deprecated `request` package with `node-fetch`.

The replacement in the newsfeed module and the e2e tests are straight forward.

In the calendar module are 3 auth methods used: base, bearer and digest. For digest the package `digest-fetch` is used additionally.

Remark: There are no tests available for bearer and digest auth, so these cases should be reviewed in detail.
closed 2021-03-13 by MichMich