MagicMirrorOrg/MagicMirror #2469: is ical still needed?

khassel, 2021-02-28:
Found no require in the code, but comments in changelog that `ical` is replaced by `node-ical`. So can this be removed?

It produces some vulnerabilities (last release is older than 1 year).

sdetweil, 2021-02-28:
yes, we left it in case there was a module that needed it

khassel, 2021-03-01:
hmm, shouldn't modules install the needed dependencies on their own?

Or do you mean such things `const ical = require("../../node_modules/ical/ical.js");`?

Was removed in v2.13.0, may we should set an end date to get rid of it?

MichMich, 2021-03-03:
I have no problem with just removing it ... a module can easily add it. And for users there is a super simple fix if it affects their installation.

PR #2473 (2021-03-13): remove ical
closes #2469 

remove oboslet package ical
closed 2021-03-13 by MichMich