MagicMirrorOrg/MagicMirror #2470: Feature request: Module config option- hide

Mystara2002, 2021-03-01:
MM provides built-in javascript functions for showing and hiding modules.  But there doesn't seem to be an easy way to actually set a module as hidden by default.

It would be handy if a standard module configuration option were available to request that a module be hidden once it has loaded.  Or in the alternative, some kind of generic configuration option to allow javascript to be freely executed once the page has loaded, so that the hide() function could be called.

There are ways of doing this (the forum suggests using MMM-AssistantMk2 to cause a notification to be sent), but this remains a cumbersome way to do something that seems relatively simple.

hangorazvan, 2021-03-01:
```
		{
			module: "name",
			position: "top_left",
			disabled: true,           // this disable (hide and not load) the module
			config: {
			}
		},
```

Mystara2002, 2021-03-01:
Hi,

Thanks.  I tried this, but it seems that if the module is disabled then it cannot be loaded at a later time (e.g. by voice or by a button).  I'm looking for something similar to 'disabled' that loads the module, but doesn't make it visible (or at the very least, makes it visible and then hide()s it).

hangorazvan, 2021-03-01:
Then the easy way still remain the js function hide() show() or make your own fuction

Mystara2002, 2021-03-01:
Correct.
So what I'm suggesting is a new configuration option to specifically call hide() on a module after that module has loaded.

Mystara2002, 2021-03-01:
Unless there's already some way to do this without modifying the module itself?

chase-cromwell, 2021-03-03:
@Mystara2002 Not totally sure I understand. Can you not just make (or edit a module) to call a js hide() on load until further action is taken? Why would there need to be a config option for this?

Mystara2002, 2021-03-03:
Editing the modules is possible, but not really practical because in my case they're a bunch (15 or so) of third party modules.

Making my own module to call js.hide() on each of the modules is possible, of course.  But I thought that having a standard configuration option of 'hidden' was such a fundamentally useful thing to have (with several requests for such standard functionality in the forums) that it really belonged in the core code, particularly since the core code itself provides hide() and show().  

A further point is that if I were to make a module that hides other modules, there would need to be some way to unambiguously refer to the instance of the module that you wanted hidden (to handle the situation where you have two instances of a module running).  I know this can be done, but it's cumbersome to say the least and not at all user-friendly.  Certainly much harder than a simple "hidden: true;" option for the module to be hidden.

MichMich, 2021-03-03:
I agree a short term solution is creating your own module. This can be very generic if you just use one common class to select the modules on boot. 

But I agree this can be a nice simple addon. feel free to send a PR which adds a `hiddenOnStartup` boolean to the module object (outside of the config).

MystaraTheGreat, 2021-03-04:
Okay, I'm a bit of a noob to MM, but I'll give this a go shortly.  There's another bug I'm trying to figure out at the moment :)

MichMich, 2021-03-13:
Merged! Closing this. Thanks for your contribution!

closed 2021-03-13 by MichMich