MagicMirrorOrg/MagicMirror #2471: The Docker image project is currently unmaintained

Legion2, 2021-03-01:
The docker image project is now over 6 months unmaintained, see https://github.com/bastilimbach/docker-MagicMirror/issues/48. @sthuber90, @khassel and I want to discuss the future of the docker image project and find new maintainers for the project. We think here (github issue) is the best place to discuss this.

khassel, 2021-03-01:
Its really strange, the owner did some merges concerning the docs, but is ignoring useful PR's since July.

MichMich, 2021-03-03:
Not sure if you already opened a forum topic for this, but if you want I can make it sticky.

khassel, 2021-03-03:
It was my idea to discuss this here, because the most technical contributors are here, not in the forum. There are several ideas to solve this problem e.g. forking the current repo or using my docker setup, but before we continue to discuss this there should be a decision from @MichMich to abandon the current one  - or if not, how to continue.

buxxi, 2021-03-06:
Couldn't we just update the docs to add the other docker image by @khassel too?
Since the documentation says this:
"The following installation methods are not maintained by the MagicMirror² core team. Use these scripts and methods at your own risk."

What are the differences between the one by @khassel and @bastilimbach?

khassel, 2021-03-06:
> What are the differences between the one by @khassel and @bastilimbach?

The images of @bastilimbach are only for the `server only` setup. Years ago I asked him to support running on raspi with electron too, but he declined. So I decided to do this on my own.

Comparison (size = compressed size on docker hub):

IMAGE     | TAG                | OS/ARCH     | RASPI WITH ELECTRON | SIZE IN MB | DESCRIPTION
------------ | ------------------ | ----------- | -------- | ------ | -------------------------------------------
bastilimbach/docker-magicmirror | latest (or v2.x.y) | linux/amd64 | no       | 499 | only `serveronly`-mode, based on debian buster
bastilimbach/docker-magicmirror | latest (or v2.x.y) | linux/arm | no       | 447 | only `serveronly`-mode, based on debian buster
bastilimbach/docker-magicmirror | latest (or v2.x.y) | linux/arm64 | no       | 489 | only `serveronly`-mode, based on debian buster
bastilimbach/docker-magicmirror | develop | linux/amd64 | no       | 509 | only `serveronly`-mode, based on debian buster
bastilimbach/docker-magicmirror | develop | linux/arm | no       | 455 | only `serveronly`-mode, based on debian buster
bastilimbach/docker-magicmirror | develop | linux/arm64 | no       | 511 | only `serveronly`-mode, based on debian buster
karsten13/magicmirror | latest (or v2.x.y) | linux/amd64 | no       | 133 | only `serveronly`-mode, based on debian buster
karsten13/magicmirror | latest (or v2.x.y) | linux/arm   | yes      | 230 | for raspberry pi, based on debian buster
karsten13/magicmirror | latest (or v2.x.y) | linux/arm64 | yes      | 255 | for raspberry pi4 64-Bit-Version, based on debian buster
karsten13/magicmirror | alpine             | linux/amd64 | no       | 72 |only `serveronly`-mode, based on alpine, smaller in size
karsten13/magicmirror | develop        | linux/amd64 | no       | 133 | only `serveronly`-mode, based on debian buster
karsten13/magicmirror | develop        | linux/arm   | yes      | 233 | for raspberry pi, based on debian buster
karsten13/magicmirror | develop        | linux/arm64 | yes      | 266 | for raspberry pi4 64-Bit-Version, based on debian buster
karsten13/magicmirror | develop_alpine | linux/amd64 | no       | 71 | only `serveronly`-mode, based on alpine, smaller in size

The `develop` images are using the `develop` branch of the MagicMirror git repository.

Beside the missing use case `raspi with electron` the image size is different. My images are stripped down to minimal size, so the downloads are faster, especially on slow devices as a raspi. Downside of the minimal approach is, that may dependencies are missing which are used by modules, e.g. python.

More questions? Just ask ...

buxxi, 2021-03-13:
I'm a bit of a beginner with docker but I tried to tinker a bit with your alpine image and eventually got it running, but as it is for now I'll stay with the unmaintained image.

It just seems so much easier to `docker pull...` and then `docker run...` than having to pull a git repo, copy files, start with a different command than the few other containers I run, enter the container for module installations etc. And I have a stateful module that I wouldn't want to lose data after a restart. It probably is the more correct way to do it your way but I mostly use it for a simplier installation/uninstallation method than building it manually more than having it sandboxed.

khassel, 2021-03-13:
I think you are talking about the use of `docker-compose`?

This has the advantage, that the `docker run ...` command is saved as a yml-File and you don't have to save this lengthly command-line somewhere else.

The `copy files` is only needed because of the different use cases which means the `docker run ...` commands are different.

I can add the corresponding `docker run ...` commands to the repo if this is helpful.

Concerning `entering the container for module installation`: This is only needed, if the tools are not on the host machine. So if you have all things on the host (`git`, `npm`, ...), you can do the installation from the host as usual.

The `stateful module` is a normal use case, there should be no problem, the `modules` folder is mapped on the host.

Running the alpine image should work out of the box with this command (I used the syntax from the bastilimbach-setup so may you have to change paths/ports for your setup):

````
docker run  -d \
    --publish 80:8080 \
    --restart always \
    --volume ~/magic_mirror/config:/opt/magic_mirror/config \
    --volume ~/magic_mirror/modules:/opt/magic_mirror/modules \
    --name magic_mirror \
    karsten13/magicmirror:alpine npm run server
````

buxxi, 2021-03-13:
Ok, thanks. Will try it later.

Should probably be in the Readme if it's a use case you're supporting :) (or at least in a "migrating from the unmaintained image" wiki page).

khassel, 2021-03-13:
yeah, the Readme is not the best, I'm always struggling with english ...

I missed 2 things:
- I decided to copy always the original `default` modules to the host because it makes no sense to use old versions which were copied by the first start of the container (may years ago). If this is unwanted you can add `--env MM_OVERRIDE_DEFAULT_MODULES="false"` to the docker run command
- the container is not running as root but as user with id=1000 (which should correspond to your normal linux user). The permissions are corrected at startup so don't wonder when in your modules/config folders permissions are changed.

So best thing would be to copy your sensitive module/config data to another location and do the tests with the copy ...

buxxi, 2021-03-13:
Documetation is boring ;)

Sounds reasonable to copy the default-folder, optimally it shouldn't even be exposed to the host. 
Maybe something can be done in MagicMirror to make that cleaner:
- env-property for path to 3rd-party-modules?
- look for modules deeper in the folder structure? 
- an optional folder `thirdparty` under `modules` which is also checked if it exists?

khassel, 2021-03-13:
problem is that the `default` folder and the 3rd-party folders are both under `modules`.

With docker we need one folder which contains the 3rd-party modules so we can mount it to the host, at the moment this is `modules` which contains also `default`.

Changing this structure e.g. using a 3rd-party-folder beside `default` would solve this problem, but would also be a breaking change for all mm users. May the env-property would be a good solution, this would point to `modules` by default but we could use another location in the docker setup.

Was your test with the alpine image successful?

buxxi, 2021-03-13:
Yes I understood the problem and all of the suggestions should be able to be done compatible without breaking for previous users.

Haven't really got the time to mess with my setup this weekend, will try the alpine image next weekend.

buxxi, 2021-03-19:
Changed my mirror to run your alpine image now, seems to be no problem for now. I'll let you know if I run into any issues.

What kind of hardcoded things did you find in the modules, relative paths to node-modules?

khassel, 2021-03-19:
> Changed my mirror to run your alpine image now, seems to be no problem for now. I'll let you know if I run into any issues.

Thanks for the feedback.

> What kind of hardcoded things did you find in the modules, relative paths to node-modules?

some modules use own html sites, e.g. MMM-Remote-Control or MMM-Strava and in these cases they have direct references in their html sites such as `<link rel="stylesheet" type="text/css" href="/modules/MMM-Strava/MMM-Strava.css">`.

So every module with such references would not work in a docker setup which would use another module path. Thats too much, I don't want to have all this support cases ...

MichMich, 2021-03-19:
In this case it would be possible @khassel ... the `/modules` endpoint is fed by the MagicMirror²'s webserver. The `/modules` endpoint can point to a different folder. It's not coupled to a location on disk.

khassel, 2021-03-19:
thanks, will test this ...

sdetweil, 2021-03-19:
this assumes all file access is via the web server, which is not the case.

as this.file() doesn't work in the helper  we all have created our own methods of getting the path to our files

buxxi, 2021-03-19:
Well, this kinda side tracked a bit... maybe create a new issue for this as a feature request instead?

As for the original issue can't we just add the image from @khassel in the documentation too and if the previous stops working for a new release then remove it if it's actually abandoned?

khassel, 2021-03-19:
> Well, this kinda side tracked a bit... maybe create a new issue for this as a feature request instead?

yes, further comments please [here](https://github.com/MichMich/MagicMirror/pull/2480) ...
 
> As for the original issue can't we just add the image from @khassel in the documentation too and if the previous stops working for a new release then remove it if it's actually abandoned?

o.k. from my side

closed 2021-06-11 by stale[bot]