MagicMirrorOrg/MagicMirror #2483: calendar date displayed wrong when distant recurring start date (1950)

sdetweil, 2021-03-16:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).
any

**Node Version**: Make sure it's version 8 or later.
14.15.0

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.
2.14

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.
the calendar code that attempts to correct for timezone shifts across timezones (and daylight savings) gets it wrong when the  recurring start date is before 2007 (or 1992, or 1952)

**Steps to Reproduce**: List the step by step process to reproduce the issue.
create a recurring full day event april 4, starting in 1952 

**Expected Results**: Describe what you expected to see.
displays april 4

**Actual Results**: Describe what you actually saw.
displays april 5

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!
default config except for calendar url

MarvinPAF, 2021-03-22:
Hi,
I see the same error, if the start year is 1940, but not, if it is 1941.

It looks like the date calculation is incorrect when on an annually repeating date 
- the start date is <1941 
- it is an all-day appointment,
The date is then moved back one day in the calendar module. 
I have attached an .ics file with sample data, their display in Thunderbird and the screen clipping from MM. 
Start-Date .ics		End-Date .ics		Date in the MM-display 
1. 19400330 		19400331 		March 31. 		error
2. 20200330 		20200331 		March 30.		OK 
The .ics file was created with Thunderbird.
I am using the current version MM from 03/19/2021 on a Raspberry Pi 4Model B Rev. 1.2 . 
The current date is March 22nd, 2021. 

[image]
[image]

sdetweil, 2021-03-22:
@MarvinPAF current version or develop?

grep version package.json
if 2.14, the do
```
git checkout develop
npm install
npm start
```

MarvinPAF, 2021-03-22:
It is the current version 2.0.1.
I have changed to develop, after install and start. There where no changes at the dates. The error is still there.
Package.json has version 2.0.1

sdetweil, 2021-03-22:
I don't see the ics file.. 
if should be a text file 
```
BEGIN:VCALENDAR
BEGIN:VEVENT
...
END:VEVENT
END:VCALENDAR
```
only need one event

MarvinPAF, 2021-03-22:
Sorry, I have had not seen the error message. .ics is not supported.
[MMM-calendar1.ics.txt](https://github.com/MichMich/MagicMirror/files/6183954/MMM-calendar1.ics.txt)

MarvinPAF, 2021-03-22:
I have changed a date from DTSTART 19400330 to DTEND 19400331 
into DTSTART 19400330T000000 to DTEND 19400330T235900
then the date is ok. May be the error is near line 335.

sdetweil, 2021-03-22:
ok, want to try this 
[calendarutils.js.txt](https://github.com/MichMich/MagicMirror/files/6184439/calendarutils.js.txt)
save and replace the modules/default/calendar/calendarutils.js  

put the dates back without the time on them  (should be the same either way now)

MarvinPAF, 2021-03-22:
I have done so and everything seems to be ok. Thank you for this fast and proper response.

sdetweil, 2021-03-22:
@MarvinPAF great, thanks. I will submit the changes.

sdetweil, 2021-04-27:
in 2.15

closed 2021-04-27 by sdetweil