MagicMirrorOrg/MagicMirror #2485: Newsfeed shows description, newsfeed displays loading regardless of hideLoading: true.

Bernard-h, 2021-03-16:
Hi my newsfeed module worked brilliantly for the last two years. I just updated my MagicMirror without editing the config.js and now I bump into problems....

Setup: Mirror has 2 newsfeed bars (one for MAJOR events in my case NL-alert and amber-alert. This bar is setup to only show for 1 hour if something serious is going on locally. The second newsfeed bar has "regular" news. The RSS still work, but I prefer my mirror nice and tidy. So just the headlines and no description. Below is the config.js and a screenshot....

It seems bar 1 does not listen to "hideLoading" anymore.
Bar 2 seems not to listen to "showDescription"=false Although I admit I added that line in the config.
Id just like the headlines and not the newsarticle description with it.

Could anyone help please? pretty please with cherry on top. Updated via git pull && npm install.

BAR1:

{
module: "newsfeed",
position: "top_bar",
config: {
hideLoading: true,
ignoreOldItems: true,
ignoreOlderThan: 6000000,
// // The config property is optional.
// // If no config is set, an example calendar is shown.
// // See 'Configuration options' for more information.

		feeds: [
			{
				title: "NL Alert / Rijnmond veiligheidsregio",
				url: "https://rss.rijnmondveilig.nl/rotterdamrijnmond/index.rss",
			},
			{
				title: "NL Alert / Rijnmond veiligheidsregio",
				url: "https://crisis.nl/algemene-informatie-van-crisisnl/",
			},
// {
// title: "Leonidas",
// url: "https://twitrss.me/twitter_user_to_rss/?user=@RKSV_Leonidas",
// },
// {
// title: "Sky News",
// url: "http://feeds.skynews.com/feeds/rss/world.xml",
// },
// {
// title: "AD Rotterdam",
// url: "https://www.ad.nl/rotterdam/rss.xml",
// },
]
},
},

BAR2

{
module: "newsfeed",
position: "top_bar",
config: {
hideLoading: true,
showDescription: false,
updateInterval: 7000,
// The config property is optional.
// If no config is set, an example calendar is shown.
// See 'Configuration options' for more information.

		feeds: [
			{
				title: "NU",
				url: "http://www.nu.nl/rss/Algemeen",
			},
			{
				title: "RTE",
				url: "https://www.rte.ie/feeds/rss/?index=/news/&limit=20",
			},
// {
// title: "Leonidas",
// url: "https://twitrss.me/twitter_user_to_rss/?user=@RKSV_Leonidas",
// },
{
title: "Sky News",
url: "http://feeds.skynews.com/feeds/rss/world.xml",
},
// {
// title: "Rijnmond",
// url: "https://wwww.rijnmond.nl/rss",
// },
]
},
},

rejas, 2021-03-17:
Hi, what version of the Mm are you using now?

Bernard-h, 2021-03-17:
Hi Rejas, I was installing MMM-pages and it seems some depreciations in dependancies caused this problem. I restored my old backup and now everything works again (except for MMMpages which I cant seem to get running , I have on open issue there). Its not a core problem of the mirror. So closing this topic. but thanks for the intended help.

closed 2021-03-17 by Bernard-h

GoudekettingRM, 2021-04-17:
I actually am facing the same problem on a clean install of MM. Cloned it earlier tonight, but the feed descriptions seem to always show, regardless of adding the `showDescription` explicitly or just using the default. 

Config: 
```js
{
  module: "newsfeed",
  position: "bottom_bar",
  config: {
    feeds: [
      {
        title: "Nieuws",
        url: "http://feeds.nos.nl/nosjournaal"
      }
    ],
    showDescription: false
  }
},
```

MMM Version: 2.15.0

rejas, 2021-04-18:
Hi @GoudekettingRM this got fixed with #2518 which is currently only in the develop branch (if you want to test that one out) and will be officially released with v2.16

GoudekettingRM, 2021-04-18:
@rejas Thanks for the quick reply. Develop seems to work perfectly. Thanks! Any idea when 2.16 will be officially released? My dad would also want this upgrade but he is not so tech savvy, so I don't want to put him on the develop branch temporarily.

MichMich, 2021-04-18:
July 1st.

GoudekettingRM, 2021-04-18:
@MichMich thanks for the quick reply! And noted.