MagicMirrorOrg/MagicMirror #2487: Async node_helper start()

MikeBishop, 2021-03-17:
If a node_helper uses an async method for start(), it appears that MagicMirror declares the system ready as soon as the function returns the initial Promise.  It would be nice if MM would collect the promises from the various modules, then `await Promise.all(...)` of them.

**Steps to Reproduce**: Use the async/await pattern within a node_helper's start() method

**Expected Results**: Initialization completes when all node_helpers are done initializing

**Actual Results**: MagicMirror reports ready while node_helper is still printing loading status to console

rejas, 2021-03-17:
Is there a module you can point to that uses async start?

MikeBishop, 2021-03-23:
I've been trying to use it in my own module, MMM-Powerwall.  I borrowed logic from MMM-RemoteControl that reads config files off the disk, and I'd already used the async file access methods elsewhere.

sdetweil, 2021-03-23:
but u only need to use async functions if u intend to use await. which  is similar to using sync calls.
there are a few different approaches to using async and await without doing it with start. 
you can start a timer, or use the notification  call from module to node helper.

rejas, 2022-10-03:
Hi @MikeBishop finally came around to look at some old issues, and whipped up a PR for your suggestion. Seems to work on my local setup but I want to get a little feedback first before I proceed.

rejas, 2022-10-29:
Fixed with #2928

MikeBishop, 2023-01-21:
So there's apparently an issue with waiting for modules that take a long time.  It looks like if initialization of the modules gets pushed out past a certain point in electron's loading process, electron will display a blank screen locally.  Accessing with a browser works fine, because everything initialized by then.

I think I have a clean repro of the issue in https://github.com/MikeBishop/MMM-asyncdeath – the synchronous start function produces this in the logs and loads normally:

```
[21.01.2023 14:55.02.692] [LOG]   Synchronous start-up -- all is well
[21.01.2023 14:55.02.693] [LOG]   Sockets connected & modules started ...
[21.01.2023 14:55.03.235] [LOG]   Launching application.
```

The async start function produces this in the logs and fails:

```
[21.01.2023 14:52.54.143] [LOG]   About to break by using await
[21.01.2023 14:52.54.884] [LOG]   Launching application.
[21.01.2023 14:52.56.512] [LOG]   Sockets connected & modules started ...
```

Note the reversed order of “Launching application” and “modules started”. Seems feasible this could also happen with a synchronous module startup that takes a long time, but I haven’t tested that.

Seems like there's an additional step needed here to delay Electron's startup until the modules are done loading.

rejas, 2023-01-21:
A quick look before bedtime makes me think the problem only occurs when starting MM with electron and not when in servermode?

rejas, 2023-01-21:
Seems like electron.js calls createWindow (when the electron app is "ready") but config isnt yet set (in the start callback at the end) due to the async nature....

MikeBishop, 2023-01-22:
> A quick look before bedtime makes me think the problem only occurs when starting MM with electron and not when in servermode?

Correct -- web browsers connecting once everything finishes startup are fine.  I think it's probably that electron launch fails (or fails to connect) and then never retries.  If so, perhaps rather than delaying electron start-up, we could kick electron to retry once the backend is ready.

rejas, 2023-01-22:
I dont like delaying stuff arbitrarily since we dont know how long some modules might take. Maybe my PR solution is enough and a little more future-proof

rejas, 2023-02-03:
@MikeBishop did you have time to check out the fixed develop branch?

PR #3021 (2023-01-26): Fix async node_helper stopping electron start
Async node_helper dont have to finish immediately in loadModules. So the start callback in the app.js with the config isnt called for some time. But the electron ready event can already be fired in the meantime.

This lead to the electron app starting but without a config (which is provded by the node_helper callback) therefor crashing. 

This PR fixes #2487 by moving the callback call out of the loadModules block, therefor the config is provided in time.

If any new async node_helper doesnt like this, we will see it :-)
commit ed90f054 (2023-01-26): Fix async node_helper stopping electron start (#3021)

Async node_helper dont have to finish immediately in loadModules. So the
start callback in the app.js with the config isnt called for some time.
But the electron ready event can already be fired in the meantime.

This lead to the electron app starting but without a config (which is
provded by the node_helper callback) therefor crashing.

This PR fixes #2487 by moving the callback call out of the loadModules
block, therefor the config is provided in time.

If any new async node_helper doesnt like this, we will see it :-)
closed 2023-04-04 by khassel