MagicMirrorOrg/MagicMirror #2490: Problems with server-mode

matze6486, 2021-03-19:
Hey,

I've got a problem with epoll in server mode with the modules "MMM-Navigate" and "MMM-Buttons".
In "non-server-mode" it works very nice

I tried the following things:
- reinstall the modules
- update npm / node.js / electron
But this doesn't work. 
Than i reinstalled the Mirror software, which doen't work, too.

Here you can see the error message:
[https://pastebin.com/WN2bYMnk](url)

Finally i reinstalled raspberry OS. But i still got the same error.

So, now i have a fresh OS, a fresh Mirror software and i installed MMM-Buttons and MMM-Navigate and need help to fix it.

Raspberry 4 B 4 GB
Raspberry OS (downloaded 19.03.2021)
MM installed via Manual installation (19.03.2021)
node v10.24.0
Here is the config [https://pastebin.com/3EspkrLX](url)

Thx in advance

khassel, 2021-03-19:
both modules needs electron, they will never work in serveronly mode.

matze6486, 2021-03-19:
allright. thx.
so, when i want to debug another modules in server mode, i only have to uncomment both modules?

khassel, 2021-03-19:
yes, you must uncomment or remove them from your `config.js`

sdetweil, 2021-03-20:
epoll is written in cc and only runs in the node helper, so why should this need electron?
node-gyp should compile in node only I think.

sdetweil, 2021-03-20:
see my update. should have posted a new message

matze6486, 2021-03-20:
I‘ll try it on monday. Thx

matze6486, 2021-03-21:
> epoll is written in cc and only runs in the node helper, so why should this need electron?
> this worked for me
> 
> from the module folder do
> 
> ```
> npm rebuild epoll --update-binary
> ```

This worked! I can Start the Mirror in server Mode!

But after doing this rebuild the non-Server Mode does not start and more :-/
The error than is the same than before the rebuild.
I can fix this this with npm install, but than i have the same Problem with the Server Mode...

sdetweil, 2021-03-21:
@matze6486 yes, it's one or the other.

electron-rebuild will fix that way. 
u could use a script to start and rebuild epoll before launching

matze6486, 2021-03-22:
Ok, Than i‘ll prefer the first Solution and uncomment the two modules, when i need to debug in Server mode
Thx a lot

sdetweil, 2021-04-01:
i think this issue can be closed

closed 2021-04-01 by matze6486