MagicMirrorOrg/MagicMirror #2497: app.js folder

shgtjgh, 2021-03-22:
## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

rejas, 2021-03-22:
Assuming that you are using the latest release from MM (and that you might not have git skills) you can just copy the contents of this file:
https://raw.githubusercontent.com/MichMich/MagicMirror/master/js/app.js

shgtjgh, 2021-03-22:
YES! thx very much!! I'm in again.

shgtjgh, 2021-03-22:
Can I ask you another question what started this mishapp in the first place?
I run PM2 in the background but the refresh time is 3 seconds. Where can I change this? I thought it was in the app.js file, but nothing happened when i changed 3000 in 10000ms in process.on SIGINT and SIGTERM

rejas, 2021-03-22:
PM2 is not part of the MM code, so there wont be anything in app.js that will change that.
From the top of my head I dont know how to change that so I suggest you ask that question in the official forums http://forum.magicmirror.builders (thats the place for support and troubleshooting).

shgtjgh, 2021-03-22:
Thx I will do that. 

Verstuurd vanaf mijn iPhone

sdetweil, 2021-03-22:
@shgtjgh what refresh time are u talking about?
mm is live, no delay. pm2 also has no delay. it is triggered by commands, file changes or some possible system interrupts... come to the forums so we can discuss your issue

rejas, 2021-03-23:
Can this be closed now @shgtjgh ?

closed 2021-04-15 by MichMich