MagicMirrorOrg/MagicMirror #2499: Some errors after latest merged commits

hangorazvan, 2021-03-23:
I found a bug in MagicMirror (maybe)
Platform: Windows 10
Node Version: 10.
MagicMirror Version: 2.15.0-develop

[2021-03-23 11:26:21.665] [ERROR] (node:3964) UnhandledPromiseRejectionWarning:
TypeError: this.broadcastEvents is not a function
    at E:\OneDrive\MagicMirror\modules\default\calendar\calendarfetcher.js:94:10
    at process._tickCallback (internal/process/next_tick.js:68:7)
[2021-03-23 11:26:21.710] [ERROR] (node:3964) UnhandledPromiseRejectionWarning:
Unhandled promise rejection. This error originated either by throwing inside of
an async function without a catch block, or by rejecting a promise which was not
 handled with .catch(). (rejection id: 8)

[2021-03-23 11:25:38.560] [ERROR] Whoops! There was an uncaught exception...
[2021-03-23 11:25:38.730] [ERROR] TypeError: self.broadcastItems is not a functi
on
    at FeedMe.<anonymous> (E:\OneDrive\MagicMirror\modules\default\newsfeed\news
feedfetcher.js:71:9)
    at FeedMe.emit (events.js:198:13)
    at SAXStream._parser.parser.emit.args [as emit] (E:\OneDrive\MagicMirror\nod
e_modules\feedme\dist\feedme.js:26:29)
    at SAXParser.SAXStream._parser.onend (E:\OneDrive\MagicMirror\node_modules\s
ax\lib\sax.js:190:10)
    at emit (E:\OneDrive\MagicMirror\node_modules\sax\lib\sax.js:624:35)
    at end (E:\OneDrive\MagicMirror\node_modules\sax\lib\sax.js:667:5)
    at SAXParser.end (E:\OneDrive\MagicMirror\node_modules\sax\lib\sax.js:154:24
)
    at SAXStream.end (E:\OneDrive\MagicMirror\node_modules\sax\lib\sax.js:248:18
)
    at XMLFeedParser._final (E:\OneDrive\MagicMirror\node_modules\feedme\dist\xm
lfeedparser.js:163:21)
    at callFinal (_stream_writable.js:617:10)
[2021-03-23 11:25:38.821] [ERROR] MagicMirror will not quit, but it might be a g
ood idea to check why this happened. Maybe no internet connection?
[2021-03-23 11:25:38.969] [ERROR] If you think this really is an issue, please o
pen an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

[2021-03-23 11:29:48.657] [ERROR] (node:7100) UnhandledPromiseRejectionWarning:
TypeError: fetcher.url is not a function
    at E:\OneDrive\MagicMirror\modules\default\newsfeed\node_helper.js:57:19
    at E:\OneDrive\MagicMirror\modules\default\newsfeed\newsfeedfetcher.js:89:5
    at process._tickCallback (internal/process/next_tick.js:68:7)
[2021-03-23 11:29:48.674] [ERROR] (node:7100) UnhandledPromiseRejectionWarning:
Unhandled promise rejection. This error originated either by throwing inside of
an async function without a catch block, or by rejecting a promise which was not
 handled with .catch(). (rejection id: 7)

MichMich, 2021-03-23:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2021-03-23 by MichMich

hangorazvan, 2021-03-23:
OK, sorry and thank you!