MagicMirrorOrg/MagicMirror #2501: loader.js is requesting none existing stylesheet custom.css

ismarslomic, 2021-03-24:
## I found a bug in MagicMirror

**Platform**:  Raspberry Pi 4, Raspberry Pi Os Buster

**Node Version**: 10

**MagicMirror Version**: 2.14.0

**Description**: `loader.js` is requesting non existing stylesheet file `custom.css`, see https://github.com/MichMich/MagicMirror/blob/3dbe8bfbbf538931e0a9c1e0c147e7d48ba42bc0/js/loader.js#L36. I see the following error message in the browser console: "Refused to apply style from http://localhost:8080/css/custom.css because its MIME type ('text/html') is not a supported stylesheet MIME type, and strict MIME checking is enabled"

**Steps to Reproduce**: Start fresh installation of Magic Mirror with default modules enabled, open Developer Console in the browser and check console and network tabs

**Expected Results**: See above

**Actual Results**: `loader.js `should not try to download a file that doesn't exist

**Configuration**: Default configuration

**Additional Notes**:

MichMich, 2021-03-24:
That's correct. The custom.css is not mandatory. Of course we could built in a check in the server to see if it is available, but other than not showing this error, this won't change anything. 

Feel free to send a PR for a clean solution.

ismarslomic, 2021-03-24:
This issue doesn't impact user experience in any way, but its just annoying seeing the error message in the browser console as developer, and I like keeping things clear. It can also be somewhat confusing error for non-developers while developing new modules. Handling this server side could be nice, or try to catch the error gracefully in `loader.js` and avoid injecting the response as Stylesheet content. That would at least remove the error message from Console (but still be there in Network).

sdetweil, 2021-03-24:
@ismarslomic if u had used my automated  (alternative) install, it would have setup an empty custom.css

it does this for two reasons
1  many new linux users don't know how to create a file
2  to eliminate the eroor you see

ismarslomic, 2021-03-24:
Good point @sdetweil! Maybe the easiest solution might be to just include empty `custom.css` file in this repo then?

sdetweil, 2021-03-24:
@ismarslomic  it used to be, but then git pull would fail because users updated it.. 

so it was removed and no longer tracked.. it doesn't hurt anything and most users never open the dev console window

ismarslomic, 2021-03-24:
I see. So by fixing one problem for users with lack of Linux and git experience another where introduced for developers. I agree that this is really not a big issue, but I used some time to understand why I got this error in the console and didn't know if it was my module or MagicMirror itself that caused it. Anyway, I can easily solve this in my own setup scripts, but I'm thinking on other developers as well.

I'm fine closing this issue if you don't share my point of view!

sdetweil, 2021-03-24:
and again, if u used the linked to 'alternative' install script, this would have been handled 
https://docs.magicmirror.builders/getting-started/installation.html#alternative-installation-methods

MichMich, 2021-03-24:
Closing since this is not a bug. Thanks for your time @ismarslomic.

ismarslomic, 2021-03-24:
I got it, but that wasn't clear for me initially. And the reason for not using automated script is because of this notification in official documentation:

>The following installation methods are not maintained by the MagicMirror² core team. Use these scripts and methods at your own risk

closed 2021-03-24 by MichMich

sdetweil, 2021-03-24:
yeh, i know.. pretty shi... considering everything I do/have done

MichMich, 2021-03-24:
@sdetweil, not sure if this issue is the appropriate issue to discuss this, but of course the "at your own risk" part is not specifically targeted at your install script. It's just good to make a distinction between core code, and third party code. Third party code always is a liability and thus this is mentioned. 

Of we should change it in "Use these scripts and methods at your own risk ... except Sam's ... Sam's scripts are safe." 😂 - But then again. If something ever goes south with your script, people come to me to say: "Hey, You said it was safe!" 🤪

So keep in mind it is nothing against your code. It's just good practice to make this distinction. I'm **very** grateful for al the time and effort you contributed.

sdetweil, 2021-03-24:
alternatively  one could have said,

These are supported in their own github repositories

without the at your own risk

MichMich, 2021-03-24:
Piping an installation script trough bash is a huge potential security risk. I think it is not a bad idea to make people aware of the fact that installation scripts come with a risk.

MichMich, 2021-03-24:
But as said, this might not be the appropriate issue to discuss this. Feel free to open an issue if you feel this issue should be discussed. Last thing I want is that you somehow feel harmed by this.

sdetweil, 2021-03-24:
your content. do whatever u want. 
I am trying to onboard new users  and reduce repeated problems in the forums.

MichMich, 2021-03-24:
And I appreciate that! I hope I convey that well.