MagicMirrorOrg/MagicMirror #2505: new and old weather module timing problem.

sdetweil, 2021-03-26:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 8 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

2.14 and 2.15-develop

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

the weather.gov provider appears to hang, for 10 minutes... before showing the forecast 
but its really a timing problem 

MM runtime calls the module to get the template data, 
and module calls thru to the provider to get data, but..   for weather.gov, the provider had to call some apis
to get the urls to call to get the data,,,  but those calls have not completed yet.
so, no forecast data is available.. and module shows loading for next update cycle, default 600,000 milliseconds (10 minutes)

**Steps to Reproduce**: List the step by step process to reproduce the issue.
see below for config

**Expected Results**: Describe what you expected to see.
forecast shown

**Actual Results**: Describe what you actually saw.
'Loading...' shown (for 10 minutes, then forecast)

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!
```
		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "weathergov",
				apiBase: 'https://api.weather.gov/points/',
				weatherEndpoint: '/forecast',
				lat: '40.7',
				lon: '-74.0',
				type: 'forecast',
			}
		}
```

phodges7727, 2021-05-12:
When I go to https://www.weather.gov/documentation/services-web-api#/, it looks like the '/points/' endpoint is now Deprecated. They have a new endpoint, "/gridpoints/{wfo}/{x},{y}/forecast". I do not have enough coding experience to rewrite the json file or else I would.

sdetweil, 2021-05-12:
your post should really be a new issue, as it is different than mine..

phodges7727, 2021-05-12:
I think your error is due to an out of date api endpoint that weather.gov has updated. They no longer support the apiBase you have shown using "/points". They now are supporting "/gridpoints/".  I do not have the coding knowledge to fix it. I just started using Magic Mirror, so I have not looked into all the js files.

sdetweil, 2021-05-12:
deprecated means don't build new apps using this..   it still works.. 

the problem in the case I discovered is looking got forecast as the ONLY report
it builds full endpoints by calling the api.   but it does this asyncronously...
so,when the module requests the forecast data, the actual api hasn't been built yet., so it fails

but it WILL succeed on the next request for the forecast, on the module updateInterval, default 10 MINUTES from now..
so it LOOKS like the module is dead.   but its not.. 

but the module doesn't keep track if it asked for data, just sends the request to the node_helper, who will get back when the data is ready.. but it can't EVER get the data NOW... 

but the helper doesn't keep track of requests BEFORE the apis are ready, so the request is lost

module side is happily waiting for data
helper side is happily waiting for requests

in updateInterval time, module will send ANOTHER request, the apis are built and it will get the data and send it back..

u can demonstrate a workaround by changing the updateInterval to a shorter time, say 1 minute, and loading will go away in 1 minute.

but you are now doing  updates every minute, and for weather this is WAY too fast...

phodges7727, 2021-05-13:
In weather.js, under the defaults, there is a value 'initialLoadDelay: 0, // 0 seconds delay'. When I changed this to anything more than 0, it works.

sdetweil, 2021-05-13:
too many darned knobs to turn.. !! lol  thanks.. 

I guess I can close this issue then

sdetweil, 2021-05-25:
closing, already a solutiuon 

'initialLoadDelay: 0  (not 0!)

closed 2021-05-25 by sdetweil