MagicMirrorOrg/MagicMirror #2507: how does MM set the require path to include the module node_modules? (per module)

sdetweil, 2021-03-29:
##  this is not a bug, it is informational 

I have a require that is failing, but I thought it would be resolved by the node_modules in my module folder....
the required submodule IS there

rejas, 2021-03-29:
Is there some code we can take a look at in github für your module?

sdetweil, 2021-03-29:
not that I am ready to share...

but during runtime, I need to use a function from one of my dependencies(hardly ever needed) and I do the typical require('library_name') 
my module npm install put this library in the node_modules of my module (as expected) 

but the require fails, unable to locate  (so it seems to be searching the root node_modules folder)

I see ways to change the path dynamically FOR ALL using code..  but that doesn't seem to be how it works in MM.. 

but all that leads me to how does it ever work?  i don't see any override for require, and I don't know how you would change it on the fly (time, promise then.)  for module context.. so I am confused..

rejas, 2021-03-29:
Normally what you describe should work: requiring your modules library in the node_module. I dont do it in my modules but looking at https://github.com/deg0nz/MMM-PublicTransportBerlin it doesnt require anything special... Maybe some typo?

sdetweil, 2021-03-29:
which node_modukes? 

no  adding the path to it works,  but worries me

rejas, 2021-03-30:
Weird, I cloned your repository, did the "npm install" and your module works.... What is the error you get?

sdetweil, 2021-03-30:
that was just an example module.  why/how does it work?

rejas, 2021-03-30:
the node js on the server side resolves that. you can go into your modules directory and start the node runtime (`node`) and issues this command in it to see where it looks:
`module.paths`

sdetweil, 2021-03-30:
but we are not in the module folder during mm execution, we are in the mm folder. if I write a file with no pathname, it is located in the mm folder

rejas, 2021-03-30:
See: https://www.freecodecamp.org/news/requiring-modules-in-node-js-everything-you-need-to-know-e7fbd119be8/
"The paths list is basically a list of node_modules directories under every directory from the current directory to the root directory."

sdetweil, 2021-03-30:
yes, but the 'current' directory is mm. , NOT the module folder
AND the current directory doesn't change during mm execution

rejas, 2021-03-30:
Mh..... I guess its some node magic or @MichMich knows it :-)

sdetweil, 2021-03-30:
@khassel  yes, it has a setPath, but what has that got to do with it?

rejas, 2021-03-31:
I think that path has nothing to do with the node_modules. Still, I can only imagine its some node.js magic that happens here....

MichMich, 2021-03-31:
> HOW does module a find ITS dependencies in node_modules(a) (which won't be in node_modules(m))
Node looks in any `node_modules` folder relative to the script that includes the `require` call.

> HOW does module b find ITS dependencies in node_modules(b) (which won't be in node_modules(a) or node_modules(m))
This is the same situation: node looks in any `node_modules` folder relative to the script that includes the `require` call.
 
> later during execution (socket notification, or setTimeout or express inbound request) , module b needs access to one of its dependencies (q) using lazy loading and does a require('q')
> this dependency is loaded in node_modules(b) and NOT in node_modules(m)
> the current folder is STILL MagicMIrror
This is the same situation: node looks in any `node_modules` folder relative to the script that includes the `require` call.

> the modules.path (require path) WAS set to MagicMirror/node_modules...
I think this only impacts which path comes first. It will alway use the default locations if it can't find the module. It will look in the node_modules folder in the same folder as the script, and will traverse to parent folders to look for other node_module folders.

Does this answer the mystery?

MichMich, 2021-03-31:
(Closed accidentally. Reopened.)

sdetweil, 2021-03-31:
ok, maybe this is the explanation  that makes sense except for a few words
https://nodejs.org/api/modules.html#modules_loading_from_node_modules_folders

quoted test
Loading from node_modules folders#
If the module identifier passed to require() is not a core module, and does not begin with '/', '../', or './', then Node.js starts at the parent directory of the current module, and adds /node_modules, and attempts to load the module from that location. Node.js will not append node_modules to a path already ending in node_modules.

If it is not found there, then it moves to the parent directory, and so on, until the root of the file system is reached.

the few words are 
starts at the parent directory of the current module

sdetweil, 2021-04-04:
i guess I answered my own question again!...

closed 2021-04-04 by sdetweil