MagicMirrorOrg/MagicMirror #2513: Proposal: CSS Custom Properties to make styling more accessible

earlman, 2021-04-02:
This solution greatly improves clarity and allows beginners to make basic aesthetic changes without having to search for selectors, go into dev tools, etc. I've read the manifesto in the docs, and I think this would be helpful for generating early positive feedback for new users getting their feet wet with programming. 

Example:
These is currently how we set the colors:
```
/* main_old.css */
body {
  color: #aaa;
}
.dimmed {
  color: #666;
}
.normal {
  color: #999;
}
.bright {
  color: #fff;
}
```
With custom properties, we would set the defaults in main.css
```
/* main.css */
:root {
    --grid-gap: 2rem;
    --color-text-body: #aaa;
    --color-text-bright: #fff;
    --color-text-normal: #999;
    --color-text-dimmed: #666;
}
body {
 color: var(--color-text-body);
} 
.dimmed {
  color: var(--color-text-dimmed);
}
.normal {
  color: var(--color-text-normal);
}
.bright {
  color: var(--color-text-bright);
}
```
and if a beginner wanted to make adjustments, they could use `custom.css.sample` (similar to how the sample config is used):
``` 
/* custom.css */
:root {
    --grid-gap: 2rem;
    --color-text-body: blue;
    --color-text-bright: #fff;
    --color-text-normal: #999;
    --color-text-dimmed: #666;
}
```
If it were up to me, I would actually just include a `custom.css` by default with some comments on how it works, but that's your call.
The custom variable for blue text in `custom.css.` overrides the defaults (#aaa) set in `main.css` like so: 
[image]

So you can see here that all a beginner would need to do was change `--color-text-body:` to start personalizing their MM. I plan to include add variables for text sizes and fonts to make it easier to tinker with all the stuff you might want to change. I'm already planning on doing some of this for my own benefit but I thought it might be helpful to the community if I shared the work.

Cheers!

rejas, 2021-04-02:
I like this idea, especially since I had that planned too (https://github.com/MichMich/MagicMirror/compare/master...rejas:custom_properties) But didnt follow it through with not enough time these days, so I'd be happy to see someone else code it  ;-)

Anyway, I am in favor of this. Like you wrote, it is easier for beginner to change the looks of MM. Plus it would be nice if Module developers would pick it up for their styles too.

sdetweil, 2021-04-02:
but it can't be custom.css or any file owned my MM in the live path   we already went thru overlaying the users custom.css file...
(sample is good, but another file people wont see)

MichMich, 2021-04-03:
It sounds like a good idea. I'm all in favor of the css changes. 

Regarding the `custom.css`: we had an issue where it was part of the repo which resulted in conflicts when users modified it. 

We ended up removing it from the repo and adding it into the `.gitignore` list. 

What could be an option though is adding a check to the startup sequence of the mirror to see if the `custom.css` file exists. And if not, generate one (which won't be tracked by git).

If someone makes PR's for this my suggestion is to separate these two changes in two separate PR's. We have to test and discuss the second one before and if we want to merge it.

rejas, 2021-04-03:
I guess the OP didnt know about the custom.css history ;-)IBut he has made a good propoal:

> and if a beginner wanted to make adjustments, they could use `custom.css.sample` (similar to how the sample config is used):

That is the more sensible idea and fits indeed nicely the other sample MM has.

rejas, 2021-04-03:
>     --grid-gap: 2rem;

Are you planning on making this customaziable too? 'cause I think font size and colors seems reasonable and managable for a first PR, but grid-gap and other padding/margins would (if at all) better fit in a later PR once we get a hang of these.

MichMich, 2021-04-03:
Agree on the grid gap. Lets take it step by step.

earlman, 2021-04-03:
Yep yep, will save layout changes for later. Not super comfortable with building a startup sequence so I’ll let somebody take that one on. Thanks all 🙂

rejas, 2021-04-03:
> Yep yep, will save layout changes for later. Not super comfortable with building a startup sequence so I’ll let somebody take that one on. Thanks all 🙂

I think you can ignore that startup sequence idea anyway, right @MichMich ?

MichMich, 2021-04-03:
Yeah that's ok.

jupadin, 2021-04-04:
> What could be an option though is adding a check to the startup sequence of the mirror to see if the `custom.css` file exists. And if not, generate one (which won't be tracked by git).

Regarding the check in the start up sequence:

- Shall we just generate an empty _custom.css_ file if there does not exist one ?
- And in case the mirror gets restarted - we then just use the (maybe empty) generated file from the previous start, right ?

Currently the console of the browser console reports that the MM can not find the _custom.css_ file, since the reason mentioned above, right ?

Once the _main.css_ uses variables it would be possible to provide a _custom.css.sample_ file - similar to the _config.js.sample_ file.

sdetweil, 2021-04-04:
if u used the automated installer it will create custom.css
https://github.com/sdetweil/MagicMirror_scripts#install-magicmirror

jupadin, 2021-04-04:
Oh okay, cool!
This might be the reason... I followed the manual installation guide given here:
https://docs.magicmirror.builders/getting-started/installation.html#manual-installation

sdetweil, 2021-04-04:
the alternative install tools  are listed after manual install

MichMich, 2021-04-15:
Merged!

closed 2021-04-15 by MichMich