MagicMirrorOrg/MagicMirror #2516: silly question. is it just the black reflection? what am i doing wrong?

jaydubelew, 2021-04-05:
I installed and ran MagicMirror successfully without any issues. So I see the weather and all the other cool stuff. 

However - the background of the screen is just black. Is the background of the screen supposed to change into something that shows the reflection, or is it that the black of the screen and the faint reflection you see is the mirror? 

The screen is a 65" vizio flatscreen. Using a Raspberry pi4. Fresh install. And again, all loaded well and without issue. Please advise. Thank you.

sdetweil, 2021-04-05:
that's just the  tv screen reflection. mm doesnt provide any background other than black. there are extension modules that show pictures.

rejas, 2021-04-05:
Or you modify your screen with a foil that turns it into a two way mirror. See the wiki entry here: https://github.com/MichMich/MagicMirror/wiki/Two-way-Mirror-Vendors

jaydubelew, 2021-04-05:
> Or you modify your screen with a foil that turns it into a two way mirror. See the wiki entry here: https://github.com/MichMich/MagicMirror/wiki/Two-way-Mirror-Vendors

- thank you! knew I was missing something!

closed 2021-04-15 by MichMich