MagicMirrorOrg/MagicMirror #2519: After upgrade MM crashes on all of my systems.

Binog, 2021-04-08:
## I'm sure it is a bug

## I'm having troubles installing or configuring MagicMirror

After a global update of all my modules and apt the system, I get the following error an all of my MM (3x)

PM2            | App [MagicMirror:0] exited with code [0] via signal [SIGINT]
PM2            | App [MagicMirror:0] starting in -fork mode-
PM2            | App [MagicMirror:0] online
0|MagicMirror  | [2021-04-08 19:16:24.600] [LOG]
0|MagicMirror  | Starting MagicMirror: v2.15.0
0|MagicMirror  | [2021-04-08 19:16:24.617] [ERROR]
0|MagicMirror  | TypeError: Cannot read property 'on' of undefined
0|MagicMirror  |     at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:93:5)
0|MagicMirror  |     at Module._compile (internal/modules/cjs/loader.js:778:30)
0|MagicMirror  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
0|MagicMirror  |     at Module.load (internal/modules/cjs/loader.js:653:32)
0|MagicMirror  |     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
0|MagicMirror  |     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
0|MagicMirror  |     at Object.<anonymous> (/usr/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23)
0|MagicMirror  |     at Module._compile (internal/modules/cjs/loader.js:778:30)
0|MagicMirror  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
0|MagicMirror  |     at Module.load (internal/modules/cjs/loader.js:653:32)
0|MagicMirror  | [2021-04-08 19:16:24.631] [ERROR]
0|MagicMirror  | Whoops! There was an uncaught exception...
0|MagicMirror  | [2021-04-08 19:16:24.632] [ERROR]
0|MagicMirror  | TypeError: Cannot read property 'on' of undefined
0|MagicMirror  |     at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:93:5)
0|MagicMirror  |     at Module._compile (internal/modules/cjs/loader.js:778:30)
0|MagicMirror  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
0|MagicMirror  |     at Module.load (internal/modules/cjs/loader.js:653:32)
0|MagicMirror  |     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
0|MagicMirror  |     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
0|MagicMirror  |     at Object.<anonymous> (/usr/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23)
0|MagicMirror  |     at Module._compile (internal/modules/cjs/loader.js:778:30)
0|MagicMirror  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
0|MagicMirror  |     at Module.load (internal/modules/cjs/loader.js:653:32)
0|MagicMirror  | [2021-04-08 19:16:24.634] [ERROR]
0|MagicMirror  | MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMirror  | [2021-04-08 19:16:24.635] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
PM2            | App [MagicMirror:0] exited with code [0] via signal [SIGINT]
PM2            | App [MagicMirror:0] starting in -fork mode-
PM2            | App [MagicMirror:0] online
0|MagicMirror  | [2021-04-08 19:16:28.263] [LOG]
0|MagicMirror  | Starting MagicMirror: v2.15.0
0|MagicMirror  | [2021-04-08 19:16:28.281] [ERROR]
0|MagicMirror  | TypeError: Cannot read property 'on' of undefined
0|MagicMirror  |     at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:93:5)
0|MagicMirror  |     at Module._compile (internal/modules/cjs/loader.js:778:30)
0|MagicMirror  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
0|MagicMirror  |     at Module.load (internal/modules/cjs/loader.js:653:32)
0|MagicMirror  |     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
0|MagicMirror  |     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
0|MagicMirror  |     at Object.<anonymous> (/usr/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23)
0|MagicMirror  |     at Module._compile (internal/modules/cjs/loader.js:778:30)
0|MagicMirror  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
0|MagicMirror  |     at Module.load (internal/modules/cjs/loader.js:653:32)
0|MagicMirror  | [2021-04-08 19:16:28.295] [ERROR]
0|MagicMirror  | Whoops! There was an uncaught exception...
0|MagicMirror  | [2021-04-08 19:16:28.296] [ERROR]
0|MagicMirror  | TypeError: Cannot read property 'on' of undefined
0|MagicMirror  |     at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:93:5)
0|MagicMirror  |     at Module._compile (internal/modules/cjs/loader.js:778:30)
0|MagicMirror  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
0|MagicMirror  |     at Module.load (internal/modules/cjs/loader.js:653:32)
0|MagicMirror  |     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
0|MagicMirror  |     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
0|MagicMirror  |     at Object.<anonymous> (/usr/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23)
0|MagicMirror  |     at Module._compile (internal/modules/cjs/loader.js:778:30)
0|MagicMirror  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
0|MagicMirror  |     at Module.load (internal/modules/cjs/loader.js:653:32)
0|MagicMirror  | [2021-04-08 19:16:28.298] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMirror  | [2021-04-08 19:16:28.299] [ERROR]
0|MagicMirror  | If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

No, I checked with default config.js 

## I found a bug in the MagicMirror Docker image

## I fo
**Platform**: Raspi 3 + 4.

**Node Version**: Make sure it's version 10 or later.

**MagicMirror Version**: latest git clone

**Description**: As stated above that happened after updating. I git clone from beginning, first start with nom works, if I do change one thing (add a modules, or change the config, or even a thing) the error above happens. clean install,. happens again.

**Expected Results**: I do expect a usual startup of MM and see the MM-Surface

**Configuration**: even with the standard config.js

closed 2021-04-08 by MichMich

sdetweil, 2021-04-08:
@Binog did u do an npm install after the git pull...

did u try with just npm start ?

Binog, 2021-04-09:
Used non install and non start. In first try after installation it worked. Second try did not work.

sdetweil, 2021-04-09:
sorry, , does this 'Used non install and non start. ' mean 

no , you did not do npm install in the module folder 
   the instructions say 'git pull && npm install'

and no, you didnt use npm start ?
what did you use to start MM? 
pm2?

Binog, 2021-04-12:
Sorry,. autocowrect... I did used npm install and npm start. And I tried pm2 start MagicMirror, and in the logs it was crashed

Binog, 2021-04-12:
ah,. it directly crashed, no module was loaeded

sdetweil, 2021-04-12:
@Binog 

what does 'crashed' mean.. can u post to the forum?
https://forum.magicmirror.builders/