MagicMirrorOrg/MagicMirror #2522: [Alert Module] Multiple Alerts doesn't show as expected

ezeholz, 2021-04-10:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Hyper-V Ubuntu

**Node Version**: v14.15.5

**MagicMirror Version**: 2.15.0

**Description**: According to issue https://github.com/Jopyth/MMM-Remote-Control/issues/256, when sending multiple alerts, the alert module can't show every alert and just hide both.

**Steps to Reproduce**: Send one alert (I tried with a 60 seconds timer), and then send the same alert.

**Expected Results**: 

[image]

**Actual Results**: 

[image]

**Configuration**: 
```
/*************** AUTO GENERATED BY REMOTE CONTROL MODULE ***************/

var config = 
{
  address: '0.0.0.0',
  electronOptions: {
    fullscreen: false
  },
  ipWhitelist: [],
  modules: [
    {
      module: 'MMM-Remote-Control',
      position: 'top_right',
      header: '',
      config: {
        showModuleApiMenu: true,
        customMenu: 'custom.json',
        secureEndpoints: false,
        classes: {
          test: {
            toggle: 'calendar'
          },
          'This is a test': {
            hide: [
              'calendar',
              'newsfeed'
            ],
            toggle: 'clock'
          },
          'this is another test': {
            show: [
              'calendar',
              'newsfeed'
            ],
            hide: [
              'clock'
            ]
          }
        }
      }
    },
    {
      module: 'updatenotification',
      position: 'top_bar',
      config: {}
    },
    {
      module: 'clock',
      position: 'top_left',
      header: '',
      config: {}
    },
    {
      module: 'calendar',
      header: 'US Holidays',
      position: 'top_left',
      config: {
        calendars: [
          {
            symbol: 'calendar-check',
            url: 'webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics'
          }
        ]
      }
    },
    {
      module: 'compliments',
      position: 'lower_third',
      classes: 'test',
      config: {}
    },
    {
      module: 'newsfeed',
      position: 'bottom_bar',
      disabled: true, // Here it's the disabled module
      config: {
        feeds: [
          {
            title: 'New York Times',
            url: 'https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml'
          }
        ]
      }
    },
    {
      module: 'alert',
      header: '',
      config: {
        welcome_message: true
      }
    }
  ],
  useHttps: false,
  httpsPrivateKey: '',
  httpsCertificate: '',
  logLevel: [
    'INFO',
    'LOG',
    'WARN',
    'ERROR'
  ]
}

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== 'undefined') {module.exports = config;}
```

**Additional Notes**: I made a fast hotfix, you could replace some lines inside https://github.com/MichMich/MagicMirror/blob/491f5aa7769a12de7f20f201de0078704c158a8d/modules/default/alert/notificationFx.js#L130

You'll have the following:
```
setTimeout(() => {
    this.ntf.classList.add("ns-hide");

    //callback
    this.options.onClose();
}, 25);
```

Select all that, and replace it with:
```
this.ntf.classList.add("ns-hide");
```

Of course that solves the issue. I tried to take the "ns-hide" part out of the function, but onClose seems to deny the normal function of the module.
I know that onClose shouldn't be dismissed, but right now it's the only way I found to bypass this bug.

Let me know if you need something else!
Cheers :D

MichMich, 2021-04-10:
Feel free to send a PR!

rejas, 2021-04-10:
Could you provide that custom.menu json so that I can take alook too? Or what do I ahve to do to get that exmaple gif recreated on my MM?

ezeholz, 2021-04-10:
@rejas Actually, the custommenu.json isn't important. You can reproduce it by using the Remote Control alert menu. I set the timer to 60, and click multiple times the send button.

rejas, 2021-04-10:
Noob question: How do I see the RemoteCOntrol alert menu?

PR #2532 (2021-04-14): Alert Module not recognizing multiple alerts
Fixes #2522

I added a condition, everything should work out normally.

onClose seems to have issues when is used with a timeout. The best way to hotfix this it's just not calling it inside the timeout.
commit 82727b82 (2021-04-14): Fix alerts

Fixes #2522
closed 2021-04-14 by ezeholz