MagicMirrorOrg/MagicMirror #2530: [Weather Module] decimalSymbol has no effect

KristjanESPERANTO, 2021-04-10:
I set the `decimalSymbol` in the config of the new weather module to `,`, but that has no effect. The values still has points. I saw that the PR #1510 should have fix that, but it doesn't work (anymore?).

I noticed this with a just freshly installed MagicMirror instance.

Or am I doing something wrong in my config?

```
    {
      module: "weather",
      position: "top_right",
      header: "Wetter in Halle (Saale)",
      config: {
        appendLocationNameToHeader: false,
        decimalSymbol: ",",
        weatherProvider: "openweathermap",
        type: "forecast",
        location: "?????????",
        locationID: "?????????",
        maxNumberOfDays: 5,
        apiKey: "?????????",
      }
    }
```

rejas, 2021-04-11:
Hi @KristjanESPERANTO I dont think you are doing somethign wrong. I took a quick look at the code and it seems to me the decimal point isnt used (yet) in the forecast part (only in the current weather part) of the module.

Maybe you want to write a fix for it? ;-)

rejas, 2021-04-11:
Nevermind, I think I can fix that if you give me a day :-)

KristjanESPERANTO, 2021-04-11:
I'm on it :slightly_smiling_face:

rejas, 2021-04-11:
Nice. So lets see if your PR fixes my tests :-D

rejas, 2021-04-17:
Did you have the time to test the latest develop branch to see if it fixes your problem for good now? So that we can close this ticekt?

KristjanESPERANTO, 2021-04-17:
Yes, with the develop branch it works fine. Thank you :slightly_smiling_face:

closed 2021-04-17 by KristjanESPERANTO