MagicMirrorOrg/MagicMirror #2539: open up discussions tab - please

B1gG, 2021-04-16:
GitHub has been supporting a discussions tab for some time now (basically issues but in a forum like a layout) and now it is also possible to transfer existing issues to discussions, to easy separate Q&A from code related topics.
I think this could be beneficial so you can have a central point to check all, also because it will enable an easy way to ask questions and discuss about features (I need that :) )

Here the how-to guide.
https://docs.github.com/en/discussions/quickstart

MichMich, 2021-04-16:
We have the existing forum for this: http://forum.magicmirror.builders

rejas, 2021-04-16:
Maybe add an explicit link to the core related part of the forum (https://forum.magicmirror.builders/category/11/core-system) to the README would be a start so that people who come to Github for technical stuff (maybe) find that?

B1gG, 2021-04-16:
Thanks, let me go over there

closed 2021-04-16 by B1gG