MagicMirrorOrg/MagicMirror #2543: Aminated icons for the standard Weather module

B1gG, 2021-04-18:
Hi,
I have been working on incorporating aminated icons to the standard Weather module, also I am proposing a mapping CSS file so we can have multiple icons set.
As per the contributions rules, I would like to check with you if this could be of interest so I can submit a PR? A fully functional PoC can be found in the Develop branch of my fork for MagicMirror, just added a config parameter `showAnimatedIcons` to the overall config and set it to true.

Cheers.

rejas, 2021-04-18:
Could you add a little image / gif to this issue to see directly what you have in mind / done in your branch? Thx in advance

B1gG, 2021-04-18:
Here you go:
[image]
I am using icons from [@basmilius](https://github.com/basmilius/weather-icons)

rejas, 2021-04-19:
I like those icons, they look nice and clean without too much animation. Adding those gets a +1 from me.

As for mapping iconsets that sounds quite nice too, but maybe that can be a sperate PR? Even one that gets merged first so that we dont have a new parameter "showAnimatedIcon" and then later a mapping parameter, but only the later one? As @michael5r stated, we shouldnt start adding too many parameter to the configs.

MichMich, 2021-04-19:
Hi @B1gG, thank you so much for your work and your proposal. 

I like te icons, but I have two issues with this change:
- The icons are pretty detailed (thin hairlines), which will probably be hard to see with a super glossy mirror in front of it. Especially since most (if not all) mirrors don't use a retina screen.
- The icons use colours. The basic design of the MagicMirror² is B&W to prevent to much noise on the mirror. So as long as the default mode isn't B&W, I won't merge such a concept. (Also, in the above concept, the arrow line of the sun rise is way to dimm to be visible behind a mirror.)

That being said: keep of your great contributions. You are on a roll! :)

B1gG, 2021-04-19:
Thanks, guys I really appreciate your feedback.
I thought it was better to ask as it is a quiet change, and as a brand, you need to keep a kind of same design (having icons in the weather but not in the calendar) Also, there are some extra icons I would like to have like the humidity and calendar (I like them because are SVG and not JS animations). I will keep this for myself for now.

Now said that @MichMich  remember you have created a (monster :) great product, stable, easy to extend (even for me I have no knowledge of Node or JS at all). I think the primary use case is what you have a "mirror", but let's considers other use cases:
- TV: I don't have any more a box on my TV, so I used the spare HDMI port to have the status of my train, bus stop, weather upcoming calendar event from my work or my daughter school, and all of that in B&W is no quite appealing.
- Digital Signage: I see a huge potential of this technology to power customs digital signage, with API interfaces calling to PowerBI reports or ServiceNow dashboards.
Perhaps what could be incorporated (v3 for example), is a Skins/Themes engine perhaps based on tailwind, and then default the B&W theme/skin but with the option to add skins and modules. What do you think?

MichMich, 2021-04-21:
Honestly, the MagicMirror² really focusses on the mirror aspects. For all other use cases it can be used, but we won't adapt the code to that. I think it's best to be great in one specific market, than mediocre in a lot of markets.

Also note that it is pretty simple to create a module which works as a skin/theme engine. We don't need to incorporate that into the core.

I agree with you that Tailwind is beautiful and super powerful. Same goes for JS frameworks like Vue. But keep in mind that if we used solutions like these, we would now be maintaining a framework that uses Grunt, Less, jQuery, etc... Those were all valid suggestions to the platform in the past 7 years. 

I agree the framework can be much more powerful using a framework like Vue combined with a CSS framework like Tailwind. But it would also make the framework way too complex for new users to get their hands dirty and to contribute to MagicMirror². 

My goal with MagicMirror² is not to built worlds most advanced SmartMirror platform. My goal is to inspire people to give software development a try and built things themselves in stead of buying a ready made products. More info about this philosophy can be found here: https://docs.magicmirror.builders/getting-started/manifesto.html

B1gG, 2021-04-21:
Thanks for taking the time to elaborate in your response, can only say I respect your position.
As the goal is to encourage newcomers and make it easy for them to customise, the focus should be on documentation and recipes to achieve from basic to complex task.
Thanks for your effort and what you have put together.

closed 2021-04-21 by B1gG

MichMich, 2021-04-21:
Thanks for your reply. Glad you understand! :)