MagicMirrorOrg/MagicMirror #2545: Proposal: Wiki Restructure

rejas, 2021-04-20:
Good morning

the WIKI could in my opinion use a little spring cleaning. All pages are on the same level without any structure. 

In a first step I grouped them in three categories: Develop / Installation / Modules.

Then I created a custom sidebar with https://github.com/adriantanasa/github-wiki-sidebar/wiki so that those groups are in a tree structure.

I wouldnt and couldnt push such a change idrectly to the wiki git sicne only @MichMich has the rights for that so I created a repo where one can see these changes: https://github.com/rejas/MagicMirrorWiki

If you have more ideas on how to freshen up the wiki please tell so. 

In my opinion it is THE place to look for modules and some tips & tricks that dont fit anywhere else. Or are there any other ideas on where to collect infos about modules?

MichMich, 2021-04-21:
I think most of the content of the Wiki can be moved to the docs website. The 3rd Party Modules list is probably the only one that needs to remain in the Wiki and Mirror Vendor List, since this is a community driven list. 

For all other aspects (installation, faq, etc) it makes more sense to put that all in one spot: the documentation page. 

Whats your thought about that?

rejas, 2021-04-21:
Yes, only the modules list (and the hw vendors) is something that has a real place here since they need to (and will) be updated often. So for the moduels maybe just seperate the one big page into subpages like finance etc, just like the voice modules already have a page?

Module development is a prime candidate for the docs indeed.

I wasnt very sure about the tips / tricks for installation, Some might already be outdated now, some might need more updates than on release day...

luwol03, 2021-04-21:
how about managing the modules in a json file in the repo and then include them into the docs with a very good customizable search which uses the json from the repo?

rejas, 2021-04-22:
That sounds nice but would require
a) someone to build something like this (could you?)
b) that MichMich would have to approve and merge an PR for every change.  Not sure if he would like that (or would delegate that work).

MichMich, 2021-04-22:
I once built a search engine for modules. The idea was that the community would help maintaining it. But in the end it was just yet one other thing for me to maintain and service. For now I prefer to just manage it in the Wiki.

rejas, 2021-04-24:
If only it would be easy to build a pipeline like https://tiny-helpers.dev/ has... Not that I wouldnt find that a tempting task but I surely dont have the time for that :-D

khassel, 2021-05-01:
I like the idea to have everything in the documentation page.

But I see no automated build process for the doc. We could leave the 3rd Party and Vendor List in a public repo and use an automated build (e.g. run weekly) for the doc stuff, where the 2 files are fetched and integrated in the doc.

MichMich, 2021-05-02:
The docs have an automated build process. As soon as I merge a change, the site will be updated.

rejas, 2021-10-09:
I'd rlike to e-open this since I might want to slowly cleanup the wiki. First target:
Deletion of https://github.com/MichMich/MagicMirror/wiki/Running-Magic-Mi
rror-on-OSX-(and-potentially-windows) 

Doesn thave any real value here, what you think @MichMich ? It would be easy to just push the delete button but I want to check back with you.

MichMich, 2021-10-09:
Yup. Go ahead. Everything that is in the documentation can be removed.

rejas, 2021-10-31:
First PR opened for some pages: https://github.com/MichMich/MagicMirror-Documentation/pull/87/files

When that gets merged following pages can be deleted in the wiki:

- https://github.com/MichMich/MagicMirror/wiki/Apple-Google-Microsoft-Calendar-Support
- https://github.com/MichMich/MagicMirror/wiki/Module-Debugging
- https://github.com/MichMich/MagicMirror/wiki/Notifications
- https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror

THe other wiki pages about on how to configure MM on linux/pi will be cleaned up later. I propose to just add a how-to for the latest raspi-os version and remove all old OS tips, sicne most users would download the latest version anyway. Any comments about that @MichMich ?

rejas, 2021-12-29:
I deleted the pages mentioned above since the PR putting them in the official documentation site is now merged.

As for the installation pages I was planning to do that during the xmas holidays but alas I dont find the time right now. Hopefully in january :-)

MichMich, 2021-12-29:
No rush! Same here. Life gets in the way. 🤷🏻‍♂️ Enjoy your holidays. 👍🏻

rejas, 2022-05-11:
Please reopen @MichMich I am not yet finished ;-)

rejas, 2022-05-11:
Could you also assing this issue to me @MichMich ?

MichMich, 2022-05-11:
Of course. 😬

khassel, 2025-05-08:
at the moment the wiki is still the place where people insert/edit/delete 3rd-party-modules so before getting rid of the wiki pages we must implement a new onboarding process for this.

KristjanESPERANTO, 2025-05-08:
I have already had several attempts to implement different approaches to replace the wiki list. But I keep coming up against the problem that I can't make it easy for both the users and us maintainers. I don't have the energy to summarize everything at the moment, but a bit of brainstorming would probably be good to list different approaches including their advantages and disadvantages.

khassel, 2025-05-08:
take your time ...