MagicMirrorOrg/MagicMirror #2547: Calendar module : reccuring annual events appear every month

lbalogh, 2021-04-21:
Hi,

Annualy reccuring events in Google Calendar appears every month in the calendar module.

See the ICS file bellow, taken directly from Google : it has 1 event that occurs 25 March of each year. But the calendar module display the event on 25 of every month.

**Platform** : Raspberry Pi 3
**MagicMirror Version** : 2.15.0

`BEGIN:VCALENDAR
PRODID:-//Google Inc//Google Calendar 70.9054//EN
VERSION:2.0
CALSCALE:GREGORIAN
METHOD:PUBLISH
X-WR-CALNAME:xxx@gmail.com
X-WR-TIMEZONE:Europe/Zurich
BEGIN:VTIMEZONE
TZID:Etc/UTC
X-LIC-LOCATION:Etc/UTC
BEGIN:STANDARD
TZOFFSETFROM:+0000
TZOFFSETTO:+0000
TZNAME:GMT
DTSTART:19700101T000000
END:STANDARD
END:VTIMEZONE
BEGIN:VEVENT
DTSTART;VALUE=DATE:20210325
DTEND;VALUE=DATE:20210326
RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTHDAY=25
DTSTAMP:20210421T154106Z
UID:zzz@google.com
CREATED:20200831T200244Z
DESCRIPTION:
LAST-MODIFIED:20200831T200244Z
LOCATION:
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:Birthday
TRANSP:OPAQUE
BEGIN:VALARM
ACTION:DISPLAY
DESCRIPTION:This is an event reminder
TRIGGER:-P0DT7H0M0S
END:VALARM
END:VEVENT
`

rejas, 2021-04-22:
I dont have much knowledge about the ics format, but I tried exporting a yearly date from my google calendar and the RRULE line was like this only:
`RRULE:FREQ=YEARLY`
and everythin looked fine on my MM.

In your ICS file the line is 
`RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;`

so probably something in there tells ical/rrule to create more events than you want. MagicMirror itself relies on that and shows these bogus dates therefor.

Could you look into your Google Calendar to see what else is in that date you export?

lbalogh, 2021-04-23:
Well, I don't have much knowledge of the ics format either, but I did some tests :

- If I create a new event in Google Calendar, it only has `RRULE:FREQ=YEARLY` and displays correctly on MM
- If I copy the "problematic" event from my girlfirend's (Google) calendar to my own (also Google) calendar, the copied version looks like `RRULE:FREQ=YEARLY;WKST=MO` and shows correctly on MM

So I don't know where `WKST=MO` comes from in this case (a bug in Google Calendar ?), but I tried to validate the `RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1` rule on some online tools, like this one : https://jakubroztocil.github.io/rrule/ and the result is correct : the event only appears 1 time each year. The `WKST=MO` seems to be ignored

lbalogh, 2021-04-23:
Again, after looking more carefully, the RRULE I posted originally (which is taken DIRECTLY from Google Calendar sharing URL) is :

`RRULE:FREQ=YEARLY;WKST=MO;INTERVAL=1;BYMONTHDAY=25`

So the problem is : `BYMONTHDAY=25`

I don't know why Google Calendar's export contain this... anyway, the simplest thing to do is asking my gf the recreate these events

rejas, 2021-04-23:
Yeah, saw that too and was wondering. Anyway, good luck with recreating it. Can this be closed then?

lbalogh, 2021-04-23:
Yes this can be closed because it's unrelated to MM

closed 2021-04-23 by lbalogh

ParaDescartar123, 2021-08-24:
Did anyone figure out what is causing this on the Google side?