MagicMirrorOrg/MagicMirror #2550: using address:"0.0.0.0" on windows causes black screen

sdetweil, 2021-04-23:
windows 10, install 2.15

start up , works
change the config.js to 
```
address:"0.0.0.0",
ipWhitelist:[].
```

restart MM.. black screen, looking at the browser dev window.. nothing is loaded
u can get to the url from another browser and it works..

u can also set the actual ip address (from ipconfig /all) and that works

this was running in a vmware workstation virtual machine

sdetweil, 2023-05-30:
Windows/Electron has some issue with 

loadUrl("http://0.0.0.0:8080")

invalid address

fix so if address:"0.0.0.0" (server is already working on all adapters)
edit js/electron.js

and change this , approx line 70 in the develop branch
```
let address = (config.address === void 0) | (config.address === "")  ? (config.address = "localhost") : config.address;
```
to this
``
let address = (config.address === void 0) | (config.address === "") | (config.address === "0.0.0.0") ? (config.address = "localhost") : config.address;
```

khassel, 2023-05-30:
I tested this with my linux/docker setup and it seems the fix has no negative impact. If you do a PR, I'd like to get a warning in the logs that says something like "config address changed from ... to ...".

khassel, 2023-07-02:
@sdetweil will you provide a PR for this? Otherwise I would create one, let me know ...

PR #3268 (2023-11-15): fix electron start loadurl on windows when address="0.0.0.0"
> - Does the pull request solve a **related** issue?
Fixes #2550 

> - What does the pull request accomplish? Use a list if needed.

changes the loadUrl to use localhost, as electron and MM are on this same system 
the mm 'server' is still listening on all adapters, including localhost
commit 247115d2 (2023-11-15): fix electron start loadurl on windows when address="0.0.0.0" (#3268)

> - Does the pull request solve a **related** issue?
Fixes #2550 

> - What does the pull request accomplish? Use a list if needed.

changes the loadUrl to use localhost, as electron and MM are on this
same system
the mm 'server' is still listening on all adapters, including localhost
closed 2024-01-01 by MichMich