MagicMirrorOrg/MagicMirror #2551: MM freezes every 20 hours

Canguingo, 2021-04-23:
Hi there,

I have MM already for a year (so pretty new with this) and since the last update I noticed that the MM started to suddenly stop. I thought was the auto start that due to the update was deleted. So I reinstalled it again, just in case. 

But now the MM freezes every 20 hours. Meaning that after 20 hours (every day around 5:28 a.m.) it freezes. when typing alt+q it takes like 5 minutes to stop and show the console. Some times I have even to unplug it to be able to stop the frozen MM. 

I know you ask a lot of info here under, but that sounds like Chinese to me.. Grateful if you could help and let me know where to find all the info you need. 

Having said this I run `npm run config:check` with the following result:

**npm ERR! code EUNSUPPORTEDPROTOCOL
npm ERR! Unsuported URL Type "config:" config:check

npm ERR! A complete log of this run can be found in:
npm ERR!  /home/pi/.npm/_logs/2021-04-23T08_11_36_983Z-debug.log**

is this relevant? 

(note: I saw 2 other posts on the same topic, but none of them solved my issue)

Many thanks in advance!
JD

 ## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 10 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

rejas, 2021-04-23:
Hi @Canguingo could you post your config.js (but beware that you dont personal inforamtion / passwords in it). And: What version of MagicMirror are you using? The latest version 2.15 or the develop brnach?

Canguingo, 2021-04-23:
Hi @rejas thanks for your quick reaction. How can I see which version am I using? I'll check in the mean time how to paste the config.js here. My pc is not connected to the MM.

Canguingo, 2021-04-23:
Hi @rejas version is 2.14.0

can't find how to update it again

Canguingo, 2021-04-23:
All right. I am trying to upgrade to the new version (i thought it was done automatically) with the code from https://github.com/sdetweil/MagicMirror_scripts 

i'm gonna check everything is all right. in the meantime i'll try to paste here the config.js

i'll come back tomorrow on this. thanks!

Canguingo, 2021-04-23:
And here the config.js (I believe I have changed all sensitive info with XXXX

/* Magic Mirror Config Sample
 *
 * By Michael Teeuw http://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information how you can configurate this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", // Address to listen on, can be:
	                      // - "localhost", "XXXX", "::1" to listen on loopback interface
	                      // - another specific IPv4/6 to listen on a specific interface
	                      // - "", "0.0.0.0", "::" to listen on any interface
	                      // Default, when address config is left out, is "localhost"
	port: XXXX,
	ipWhitelist: ["XXXX", "::ffff:XXXX", "::1"], // Set [] to allow all IP addresses
	                                                       

	language: "nl",
	timeFormat: 24,
	units: "metric",

	modules: [
		{
        		module: 'MMM-ModuleScheduler'
		},
		 
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},

		{
			module: "currentweather",
			position: "top_right",
			config: {
				location: "Maarssen",
				locationID: "",  //ID from http://bulk.openweathermap.org/sample/; unzip the gz file and find your city
				appid: "XXX"
			}
		},
//		{
//			module: "compliments",
//			position: "lower_third"
//		},
		{
			module: "calendar",
			header: "Agenda",
			position: "top_left",
			showEnd: "False", 
			config: {
				calendars: [
					
					
					{
						symbol: "calendar",
						showEnd: false,
						url: "XXXX",
					},
{
						symbol: "calendar-check-o ",
						showEnd: false,
						url: "XXXX"
					}
					],
			}
		},
//		{
//            module: 'MMM-Buienradar',
//            position: 'top_right',
//            config: {
//              lat: XXXX,
//              lon: XXXX,
//              forecast: true,
//              zoom: 4,
//              interval: 10
//		}
//        } 
//	{
//    module: "MMM-EARTH",
 //   position: "bottom_center",
 //   config: {
 //       mode: "Natural",
 //       rotateInterval: 2000,
 //       MaxWidth: "50%",
 //       MaxHeight: "50%",
 //   }
//}	
		
		
	]

};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

luwol03, 2021-04-23:
could you please put this into a codeblock and maybe even in a summary tag?

Canguingo, 2021-04-26:
I have been testing the MM and it seems the change in logs and updating config has solved the issue. Estrange I did not get a warning of not having the last version. 
Since I haven't had this issue in the last 36 hours I will close this issue. Many thanks all have reacted so quickly. I helped me a lot.

closed 2021-04-26 by Canguingo