MagicMirrorOrg/MagicMirror #2553: Black screen (Modules not appearing)

ahmed-asaleh, 2021-04-24:
Hello, I have installed my Magic Mirror and it has been working for a while without any issues
But now, the screen is just black and the modules are not appearing.
The MagicMirror tab used to have the name MagicMirror on the top of the screen, but now it is called electron(i don't know why)
Can anyone please tell me what to do? I am still a beginner, so please tell me what to do step by step.

Thanks in advance.

sdetweil, 2021-04-24:
this usually means there is an erorr..

how do you start MM?  thru npm start, or pm2?  

if pm2, then the pm2 logs --lines=100 command should show the output

minggaofu, 2021-06-28:
I am having issues as well. first install works (calendar and news don't load) reboot only get black screen. 
npm Start shows below
28.06.2021 22:02.25.572] [LOG]   Starting MagicMirror: v2.15.0
[28.06.2021 22:02.25.581] [LOG]   Loading config ...
[28.06.2021 22:02.25.598] [LOG]   Loading module helpers ...
[28.06.2021 22:02.25.603] [LOG]   No helper found for module: alert.
[28.06.2021 22:02.26.072] [LOG]   Initializing new module helper ...
[28.06.2021 22:02.26.074] [LOG]   Module helper loaded: updatenotification
[28.06.2021 22:02.26.078] [LOG]   No helper found for module: clock.
[28.06.2021 22:02.27.430] [LOG]   Initializing new module helper ...
[28.06.2021 22:02.27.432] [LOG]   Module helper loaded: calendar
[28.06.2021 22:02.27.435] [LOG]   No helper found for module: compliments.
[28.06.2021 22:02.27.438] [LOG]   No helper found for module: weather.
[28.06.2021 22:02.27.448] [ERROR] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'valid-url'
Require stack:
- /home/pi/MagicMirror/modules/default/newsfeed/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[28.06.2021 22:02.27.450] [LOG]   Loading module helpers ...
[28.06.2021 22:02.27.452] [LOG]   No helper found for module: alert.
[28.06.2021 22:02.27.453] [LOG]   Initializing new module helper ...
[28.06.2021 22:02.27.455] [LOG]   Module helper loaded: updatenotification
[28.06.2021 22:02.27.456] [LOG]   No helper found for module: clock.
[28.06.2021 22:02.27.458] [LOG]   Initializing new module helper ...
[28.06.2021 22:02.27.459] [LOG]   Module helper loaded: calendar
[28.06.2021 22:02.27.461] [LOG]   No helper found for module: compliments.
[28.06.2021 22:02.27.464] [LOG]   No helper found for module: weather.
[28.06.2021 22:02.27.468] [ERROR] App threw an error during load
[28.06.2021 22:02.27.471] [ERROR] Error: Cannot find module 'valid-url'
Require stack:
- /home/pi/MagicMirror/modules/default/newsfeed/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:972:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:249:921)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:848:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12633)
    at Module.require (internal/modules/cjs/loader.js:1032:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/newsfeed/node_helper.js:9:16)
    at Module._compile (internal/modules/cjs/loader.js:1152:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1173:10)
[28.06.2021 22:02.27.474] [ERROR] Whoops! There was an uncaught exception...
[28.06.2021 22:02.27.490] [ERROR] Error: Cannot find module 'valid-url'
Require stack:
- /home/pi/MagicMirror/modules/default/newsfeed/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:972:15)
    at Function.n._resolveFilename (electron/js2c/browser_init.js:249:921)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:848:27)
    at Function.f._load (electron/js2c/asar_bundle.js:5:12633)
    at Module.require (internal/modules/cjs/loader.js:1032:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/newsfeed/node_helper.js:9:16)
    at Module._compile (internal/modules/cjs/loader.js:1152:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1173:10) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/default/newsfeed/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[28.06.2021 22:02.27.497] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[28.06.2021 22:02.27.499] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[28.06.2021 22:02.28.265] [LOG]   Launching application.

sdetweil, 2021-06-28:
@minggaofu   

Cannot find module 'valid-url'

yes, this was removed in 2.15, and some module uses it, but didn't document it as a dependency

workaround
```
cd ~/MagicMirror
npm install valid-url
```

MichMich, 2021-06-29:
@ahmed-asaleh Does this solve your problem? If so I can close this issue. Thanks.

minggaofu, 2021-06-29:
Yes Thanks for your help

MichMich, 2021-06-29:
Thanks. Closed. @sdetweil Thanks again for your great support!

closed 2021-06-29 by MichMich

Tv-igtmomsp, 2021-06-29:
Indeed Thank you!!