MagicMirrorOrg/MagicMirror #2555: end of life of node v10

khassel, 2021-04-25:
From the forum: https://forum.magicmirror.builders/topic/14987/official-recommended-node-js-version-for-mm
Node Releases: https://nodejs.org/en/about/releases/

End of live of node v10 is 2021-04-30.

In [the documentation](https://docs.magicmirror.builders/getting-started/installation.html#manual-installation) still node v10 is mentioned as default node installation.

This should be updated, my suggestion would be v14 (I'm already using this version a long time in my docker setup).

The git workflow is using v10/v12/v14, may we should move here to v12/v14/v16?

rejas, 2021-04-26:
Agree. Although it would be good to know what popular distrubutions like the raspiOS will ship from then on as the default node environment. If we increase the node version too much people would need to install node themselves which might lead to more questions /issues :-)

khassel, 2021-04-26:
AFAIS there is no default node environment in raspberry pi os, you have always to install node by yourself or use the script of @sdetweil (which is using v10 at the moment).

The lowest active version of node is v14.

sdetweil, 2021-04-26:
i will change my script to whatever we decide on.. easy 1 variable change

sdetweil, 2021-04-26:
i can provide  a script to upgrade node version to whatever we decide

MichMich, 2021-04-26:
v14 is the current LTS. So it makes sense to switch to v14. I think it's also a good idea to update the test suite to use 12, 14 & 16.

rejas, 2021-04-26:
+1 from me for v14 too

khassel, 2021-04-28:
Script of @sdetweil is also updated to v14 so I will close this issue.

closed 2021-04-28 by khassel