MagicMirrorOrg/MagicMirror #2558: Update fails to install electron-chromedriver@11.0.0

camaro4life18, 2021-04-27:
When I try to update to the latest version, Raspberry Pi 4 on latest Raspbian, I am getting this error...

pi@magicmirror:~/MagicMirror $ git pull && npm install
Already up to date.

> electron-chromedriver@11.0.0 install /home/pi/MagicMirror/node_modules/electron-chromedriver
> node ./download-chromedriver.js

/home/pi/MagicMirror/node_modules/fs-extra/lib/index.js:23
SVGTestsImpl.attributeRegistry = new Map([
                                         ^

SyntaxError: Unexpected end of input
    at Module._compile (internal/modules/cjs/loader.js:723:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/@electron/get/dist/cjs/utils.js:4:12)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron-chromedriver@11.0.0 install: `node ./download-chromedriver.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the electron-chromedriver@11.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-04-27T21_34_52_730Z-debug.log

sdetweil, 2021-04-27:
can u tell us which version of node and npm u have installed

node -v 
npm -v

camaro4life18, 2021-04-27:
***@***.***:~ $ npm -v
6.14.12
***@***.***:~ $ node -v
v10.24.1
***@***.***:~ $

sdetweil, 2021-04-27:
chromedriver is really only for testing

can u try this

```
cd ~/MagicMirror
npm install --only=prod
```
this will skip all the development/test stuff

camaro4life18, 2021-04-28:
Well after running that command, it gets stuck in an infinite loop with the below error. I followed its recommendations and deleted the electron folder, which I had to run the regular npm install command to get it to re-install electron, which then failed on the chromium node. I'm going to mv the whole MagicMirror folder and try a clean install.

`Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
    at getElectronPath (/home/pi/MagicMirror/node_modules/electron/index.js:14:11)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/index.js:18:18)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/cli.js:3:16)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.15.0 start: `DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-04-28T00_39_26_589Z-debug.log
/home/pi/MagicMirror/node_modules/electron/index.js:14
    throw new Error('Electron failed to install correctly, please delete node_modules/electron and try installing again')
    ^
`

camaro4life18, 2021-04-28:
Well, something had to be corrupted with my npm nodes, especially electron. Doing a clean checkout and "npm install" worked with no error. I reinstalled all my modules and only had 1 issue missing the "valid-url" node for MMM-Calendar-Ext, but doing npm install valid-url fixed the issue.

So back up and running now. I can keep my backup of the old install if you want me to do any more troubleshooting, otherwise I'll close this issue.

closed 2021-04-28 by camaro4life18