MagicMirrorOrg/MagicMirror #2571: add windows postinstall script

sdetweil, 2021-05-24:
it would be nice to resolve the issues with native windows install

the post install script won't run, but I finally figured out a reliable way 

scripts launch a shell  (sh or cmd.exe)  and a single postinstall file won't work

BUT   NODE runs..
"postinstall":"node somejsfile.js"
and u can pass parms

so I wrote a little launcher 

"anyscript":"node  execscript.js  _linux_.sh:_windows_.cmd"
so, on any script, this will launch node execscript.js  and pass the string  "_linux_.sh:_windows_.cmd"

and then execscript will launch the _linux_.sh script on not windows platforms and the windows script (if supplied) on windows

i shortened the execscript.js name to es.js   fewer chars...

SO, we could fix start too 
"start": "node es.js 'DISPLAY=\"${DISPLAY:=:0}\" ./node_modules/.bin/electron js/electron.js:node_modules/.bin/electron js/electron.js'"

 or pass a script (runstart.sh or runstart.cmd)

MichMich, 2021-06-29:
Sorry, I don't fully understand the issue or what's needs to be solved. Isn't the post install issues solved by a newer version of NPM?

sdetweil, 2021-09-01:
sorry, didn't see this response

so far not

you cannot run a sh file on windows
you cannot do any logic in the package.json script clause that isn't linux based
u cannot run a windows program on windows (doesn't work)
u cannot use the windows path with backslash in the text for the script action

but you CAN run node, and some js file

SO, if the js file can determine platform *and it can*, then it could launch a platform specific script/process 
(which could be passed as parameters to the node js file)

closed 2022-04-27 by stale[bot]