MagicMirrorOrg/MagicMirror #2582: two modules use the same library name, but at different levels, how to resolve

sdetweil, 2021-06-05:
my MMM-Config and MMM_WifiPassword both use  a QRCode

i get mine from the nodejs catalog 
he has his copied from somewhere

whichever is loaded last  wins...
but the other module doesn't work

I changed his code to be explict path like mine, but no help 
so its the function names INSIDE the file that wins.. 

I could use his lib, but no guaranty it will continue to work.. 

is there a known solution to this problem?  can I somehow scope my getscripts() response to my module only?

sdetweil, 2021-06-05:
looks like there are different ways IF you control the loading process (and so can replace the name with a local instance)
but I don't ...

sdetweil, 2021-06-08:
anyone have any ideas how to solve this generically?

MichMich, 2021-06-08:
The server side is easy. But the client side is difficult because we don't use something like webpack.

sdetweil, 2021-06-08:
but there has GOT to be something...   this isn't a new problem with MM I don't think

MichMich, 2021-06-08:
I haven't come across it before. Is it. Prefect that it is a client side framework?

sdetweil, 2021-06-08:
sorry, don't understand the last sentence.. 

it is a framework for homolgating content from disparate sources. 
how should I protect my code from old, abandoned modules?

MichMich, 2021-06-08:
Haven't tried it, but maybe you can import the lib as a module.

sdetweil, 2021-06-08:
yes I could use it in my node_helper,  I already have it as a dependency. 

but why should I have to redesign my code.. grrrr

https://www.npmjs.com/package/qrcode#tofilepath-text-options-cberror

https://stackoverflow.com/questions/13938686/can-i-load-a-local-file-into-an-html-canvas-element

MichMich, 2021-06-08:
Dealing with edge cases is the fun of programming. 😂

closed 2021-06-08 by sdetweil