MagicMirrorOrg/MagicMirror #2585: module configuration with external config.js or json file (for 3rd party)

bugsounet, 2021-06-09:
I purpose to create a news feature for MM

What about `configFile: "My_module_config.json"`
`My_module_config.json` will be located in the module directory

naturaly, we can also use `config:{}` but not at the same time

What do you this about this ?
Personaly, i think i could be better for more visibility
and any user will be not `lost` in the config.js file

sdetweil, 2021-06-09:
my MMM-Config goes sort of in this direction.. one can save and customize the information for the modules form presentation.
drops downs, date,  name,  etc fields. 
and then generates the config file.. user never sees it.  if there is a schema.json file in the module folder my module will use it. (similar to your idea)   

MM doesn't change
for abandoned modules, i will collect customized schemas from users and use those when needed

the users get form based admin to their Magic Mirror

bugsounet, 2021-06-09:
lol ... I do it too with MMM-Setup too but i have to done it ;)

sdetweil, 2021-06-09:
I don't see that on github,, do you do this for every/any module? without their participation?

bugsounet, 2021-06-09:
naturally not for all modules ... only for mine !
and Naturally too ... I don't ads it ;) [module config creator with user interaction i config.js]

But ... it's not the main subject

closed 2021-09-06 by stale[bot]