MagicMirrorOrg/MagicMirror #2587: [Raspberry PI 3B] MagicMirror ends after few hours

tuxrinku, 2021-06-10:
## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 3B with latest Raspberry Pi OS

**Node Version**: 16.3.0

**MagicMirror Version**: 2.15

**Description**: MagicMirror runs fine for a few hours, then exits with the following error: `../../third_party/tcmalloc/chromium/src/tcmalloc.cc:1462] Pointer is not pointing to the start of a span`
`/home/pi/Downloads/MagicMirror/node_modules/electron/dist/electron exited with signal SIGSEGV`

**Configuration**: [config.js.txt](https://github.com/MichMich/MagicMirror/files/6632030/config.js.txt)

**Additional Notes**: Only added the MMM-BackgroundSlideshow module on top of the default ones.

MichMich, 2021-06-29:
Could you run your mirror without the 3rd-party module (MMM-BackgroundSlideshow) to see if it suffers from te same issue?

Tv-igtmomsp, 2021-06-29:
https://github.com/MichMich/MagicMirror/issues/2587

MichMich, 2021-06-29:
@Tv-igtmomsp It looks like you are spamming the comments. Any reason you tagged this issue? 🤔

tuxrinku, 2021-07-02:
Yes it was the culprit, there's no problem running MagicMirror without that module.

MichMich, 2021-07-02:
Good to hear! I suggest you open an issue in that repo. Can I close this issue?

tuxrinku, 2021-07-02:
Sure thing

closed 2021-07-02 by tuxrinku