MagicMirrorOrg/MagicMirror #2590: Calendar module - Hidden private events

borishalleux, 2021-06-12:
Hello,

With the default calendar module, when you hide private events, they are still counted in a maximum number of events to display.
personally i think they shouldn’t be displayed to have a more complete view of the calendar, right?

Sorry for my English, Google translation, I only speak French.

rejas, 2021-06-13:
hi @borishalleux I agree, it would be more logial to not count the private events. 
Would you like to probide a PR for this topic or is this just a feature request :-) ?
Cheers

borishalleux, 2021-06-15:
Hello,
For me this is not a feature but a correction.
But I don't know enough about GitHub to know what to do with my post next.

rejas, 2021-06-24:
Maybe @MichMich can add the "PR welcome" label to this issue (if he agrees with the content) so that other might add it?

MichMich, 2021-06-24:
Done!

relm923, 2021-06-30:
Based on the current code (https://github.com/MichMich/MagicMirror/blob/master/modules/default/calendar/calendar.js#L491 and https://github.com/MichMich/MagicMirror/blob/master/modules/default/calendar/calendar.js#L573) its not counting private events when calculating `maximumEntries`.

@borishalleux can you post your Calendar config and some more details about your events? I'm struggling to reproduce your edge case

khassel, 2023-09-27:
I've done some testing and I also think this is fixed in the current code

```js
				if (this.config.hidePrivate && event.class === "PRIVATE") {
					// do not add the current event, skip it
					continue;
				}
```

So closing ...

closed 2023-09-27 by khassel