MagicMirrorOrg/MagicMirror #2596: MMM-MonthlyCalendar private events not showing up

FlorjanKlancar, 2021-06-15:
I have a problem with MMM-MonthlyCalendar model. Widget is displaying events that are in public google calendar but can’t display events that are linked to my private google account calendar…
Is there any reason why? I give him the url just like the default magic mirror calendar view (it works there)?

Btw ignore the overlaying modules the screen is supposed to be a larger screen.

Config:
{
                        module: "MMM-MonthlyCalendar",
                        position: "bottom_bar",
                        config: {
                                updatesEvery: 30,
                                weeksInFuture: 5,
                                calendars: [
                                {
                                        url: "<filename>",
                                        priority: 1
                                },
                                {
                                        url: "<filename>",
                                        priority: 2
                                },

                                ]
                        }
                },
[image]

rejas, 2021-06-20:
Hi @FlorjanKlancar since this is not one of the default modules please refer to http://forum.magicmirror.builders for support and troubleshooting.

rejas, 2021-06-20:
Ah I see you already did: https://forum.magicmirror.builders/topic/15132/mmm-monthlycalendar?page=1
so this issue can be closed then?

closed 2021-06-20 by MichMich