MagicMirrorOrg/MagicMirror #2598: Calendar module not displaying iCloud and Google events

jeremycjang, 2021-06-16:
## I found a bug in MagicMirror

**Platform**: Raspberry Pi 4

**Node Version**: v14.16.1

**MagicMirror Version**: 2.15

**Description**: I've attempted to add personal calendars to my config.js , but they aren't displaying when the program runs.

One is iCloud (publicly shared) and one is Google Calendar (private calendar). 

Both links are valid because I can load them in Windows Outlook and see events no problem. The Gmail Calendar was even working properly before.

I've left the default US holiday calendar in and it IS showing those events, so the module itself is parsing the links but ignoring them.

If I comment out/remove the default calendar, "No upcoming events" is displayed (incorrectly, there are upcoming events on my calendars).

**Expected Results**:  All events showing on calendar

**Actual Results**: MM ignoring valid calendar links besides default calendarlabs link

**Configuration**: 

```

		{
			module: "calendar",
			header: "Upcoming Events",
			position: "top_left",
			config: {
				calendars: [
					{	// iCloudCalendar
						symbol: "calendar-check",
						url: "webcal://p66-caldav.icloud.com/published/2/[redacted].ics",
						// Gmail Calendar
						symbol: "calendar-check",
						url: "https://calendar.google.com/calendar/ical/[redeacted]/private-[redacted]/basic.ics",
						//Basic US Holiday Calendar
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"					}
				]
			}
		},
```

sdetweil, 2021-06-16:
each cal url needs its own block

{
url
},
{
url
}

recall the per cal stuff goes in there too

jeremycjang, 2021-06-16:
> each cal url needs its own block
> 
> {
> url
> },
> {
> url
> }
> 
> recall the per cal stuff goes in there too

you're the best; thanks so much!

closed 2021-06-16 by jeremycjang