MagicMirrorOrg/MagicMirror #2606: Google Assistant not Responding using hot word

prabashanash, 2021-06-26:
I hope someone can help me with this problem. installed magic mirror and added MMM-GoogleAssistant and MMM-Snowboy
google assistant is not responding with the hot word : jarvis

added all in config > my config below
----------------------------------------------------------
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
    address: "localhost", // Address to listen on, can be:
    // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
    // - another specific IPv4/6 to listen on a specific interface
    // - "0.0.0.0", "::" to listen on any interface
    // Default, when address config is left out or empty, is "localhost"
    electronOptions: {
        webPreferences: {
            webviewTag: true,
        },
    },
    port: 8080,
    basePath: "/", // The URL path where MagicMirror is hosted. If you are using a Reverse proxy
    // you must set the sub path here. basePath must end with a /
    ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses
    // or add a specific IPv4 of 192.168.1.5 :
    // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
    // or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
    // ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

    useHttps: false, // Support HTTPS or not, default "false" will use HTTP
    httpsPrivateKey: "", // HTTPS private key path, only require when useHttps is true
    httpsCertificate: "", // HTTPS Certificate path, only require when useHttps is true

    language: "en",
    locale: "en-US",
    logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
    timeFormat: 24,
    units: "metric",
    // serverOnly:  true/false/"local" ,
    // local for armv6l processors, default
    //   starts serveronly and then starts chrome browser
    // false, default for all NON-armv6l devices
    // true, force serveronly mode, because you want to.. no UI on this device

    modules: [
        {
            module: "alert",
        },
        {
            module: "updatenotification",
            position: "top_bar",
        },
        {
            module: "clock",
            position: "top_left",
        },
        {
            module: "calendar",
            header: "Holidays & Events",
            position: "top_left",
            config: {
                calendars: [
                    {
                        symbol: "calendar-check",
[... 452 lines cut ...]
    name: '@bugsounet/node-lpcm16',
    installed: '2.0.1',
    latest: '2.0.1'
  },
  { name: '@bugsounet/cvlc', installed: '1.0.4', latest: '1.0.4' },
  {
    name: '@bugsounet/google-photos',
    installed: '1.0.0',
    latest: '1.0.0'
  },
  { name: '@bugsounet/systemd', installed: '1.0.0', latest: '1.0.0' }
]
[26.06.2021 22:51.18.806] [LOG]   [NPM] Check... MMM-GoogleAssistant
[26.06.2021 22:51.18.807] [LOG]   [NPM] No update needed for MMM-GoogleAssistant
[26.06.2021 22:51.20.238] [INFO]  Newsfeed-Fetcher: Broadcasting 20 items.
[26.06.2021 22:51.21.655] [INFO]  Calendar-Fetcher: Broadcasting 1 events.
[26.06.2021 22:51.21.807] [INFO]  Calendar-Fetcher: Broadcasting 7 events.
[26.06.2021 22:56.22.447] [INFO]  Newsfeed-Fetcher: Broadcasting 20 items.
[26.06.2021 22:56.23.631] [INFO]  Calendar-Fetcher: Broadcasting 1 events.
[26.06.2021 22:56.23.772] [INFO]  Calendar-Fetcher: Broadcasting 7 events.

Any further hints / tips ?

Kind regards,
Praba

sdetweil, 2021-06-26:
sorry, the author of that module doesn't provide support here..

you need to go to 
http://forum.bugsounet.fr

prabashanash, 2021-06-26:
Thanks @sdetweil

closed 2021-06-26 by prabashanash