MagicMirrorOrg/MagicMirror #2608: After Update to 2.1.6 no start

TheBassfly, 2021-07-01:
Hey there

After updating my Magic Mirror today, it won't start anymore.

When I tried to start it via NPM Start, I got the message:

sh: 1: ./node_modules/.bin/electron: not found

So I tried to install Electron via:

npm install electron@3.0.10

After this, I got this error message:

[01.07.2021 16:37.27.915] [ERROR] App threw an error during load
[01.07.2021 16:37.27.946] [ERROR] TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:36:38
    at /home/pi/MagicMirror/js/logger.js:19:20
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:24:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:52:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
A JavaScript error occurred in the main process
Uncaught Exception:
TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:36:38
    at /home/pi/MagicMirror/js/logger.js:19:20
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:24:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:52:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)

What is it, that I'm doing wrong?

Thanks

khassel, 2021-07-01:
current version is 2.16.0, not 2.1.6

MichMich, 2021-07-01:
What steps did you take to upgrade?

TheBassfly, 2021-07-01:
Gut pull & npm install

There was an error with th git pull, so I used git reset --hard and then it worked.

MichMich, 2021-07-01:
Good to hear! Can I close this issue?

TheBassfly, 2021-07-01:
No this was the answer to your question how I did the upgrade.

MichMich, 2021-07-01:
Ah sorry. Thought you fixed it. What version of Node are you running?

TheBassfly, 2021-07-01:
Node 15.14.0
Npm 7.11.1

MichMich, 2021-07-01:
Are you running any 3rd party modules?

TheBassfly, 2021-07-01:
Yes
Calenderext2 and a open weather module.

MichMich, 2021-07-01:
Try disabling those modules to see if that helps.

TheBassfly, 2021-07-03:
Are you sure with this? This means for me to del around 80 lines in the config.js. And if this is the problem, the Whole MM would be useless, because all these third party Modules re essential and there no alternatives in the moment.

KristjanESPERANTO, 2021-07-03:
This is just a test to find out what the problem is. Save your config.js before, so you can easily undo the changes afterwards.

MichMich, 2021-07-03:
What @KristjanESPERANTO says. It's just to pinpoint the issue. 

You can add comment blocks to make it easier.

khassel, 2021-07-03:
> npm install electron@3.0.10

we are on v11

TheBassfly, 2021-07-04:
> > npm install electron@3.0.10
> 
> we are on v11

Thanks for this hint. I updated to v11 and when I start MM again, it show this:

> magicmirror@2.16.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[04.07.2021 12:26.22.738] [ERROR] App threw an error during load
[04.07.2021 12:26.22.750] [ERROR] TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:36:38
    at /home/pi/MagicMirror/js/logger.js:19:20
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:24:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:52:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
A JavaScript error occurred in the main process
Uncaught Exception:
TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:36:38
    at /home/pi/MagicMirror/js/logger.js:19:20
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:24:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:52:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)

TheBassfly, 2021-07-04:
> What @KristjanESPERANTO says. It's just to pinpoint the issue.
> 
> You can add comment blocks to make it easier.

I will try in the next days as I'm niche 24/7 working the MM :). I Keep you updated.

TheBassfly, 2021-07-05:
> Try disabling those modules to see if that helps.

So, I removed all the third party modules in the config and this happened after "nom start".

> magicmirror@2.16.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[05.07.2021 11:30.20.021] [ERROR] App threw an error during load
[05.07.2021 11:30.20.077] [ERROR] TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:36:38
    at /home/pi/MagicMirror/js/logger.js:19:20
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:24:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:52:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
A JavaScript error occurred in the main process
Uncaught Exception:
TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:36:38
    at /home/pi/MagicMirror/js/logger.js:19:20
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:24:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:52:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)

MichMich, 2021-07-05:
Try removing the node_modules folder and then run `npm install` again.

TheBassfly, 2021-07-05:
This is the actual Config:

> Try removing the node_modules folder and then run `npm install` again.

Did this but it ended up in the same messages like above.

MichMich, 2021-07-05:
I don't have a clue what this can be. Can you give a bit more info about your environment?

Os, Version, Hardware, etc.

MichMich, 2021-07-05:
That should not be a problem. The only thing I can think of is a node version issue. It might be worth trying 14 or 16. If you install node `n` trying this is easy.

TheBassfly, 2021-07-05:
> That should not be a problem. The only thing I can think of is a node version issue. It might be worth trying 14 or 16. If you install node `n` trying this is easy.

I will give it a try tomorrow.
I just left the versions because some of the homebridge plug-ins awaited lower versions. Let's try it tomorrow.

rejas, 2021-07-05:
This sounds like this here: https://forum.magicmirror.builders/topic/13362/magic-mirror-2-won-t-start

TheBassfly, 2021-07-06:
> This sounds like this here: https://forum.magicmirror.builders/topic/13362/magic-mirror-2-won-t-start

Nope
As I'm coming from v. 2.1.5 I did this solution with Git Pull + npm install.

TheBassfly, 2021-07-06:
> That should not be a problem. The only thing I can think of is a node version issue. It might be worth trying 14 or 16. If you install node `n` trying this is easy.

Okay. I made a switch to Node 14.17.3 and NPM 6.14.13
And made an apt-get Upgrade for the System.

At the Ende same error again

sdetweil, 2021-07-06:
@TheBassfly you also have to remove node_modules folder and package-lock.json 
and then npm install again

npm7 is the culprit. not passing env correctly. 
I see DISPLAY  missing if I use a script to start mm.   npm start in the mm folder is ok

TheBassfly, 2021-07-06:
> @TheBassfly you also have to remove node_modules folder and package-lock.json
> and then npm install again
> 
> npm7 is the culprit. not passing env correctly.
> I see DISPLAY missing if I use a script to start mm. npm start in the mm folder is ok

Thanks.
I did it like this but ended up in the same issue. Maybe I have to delete Magic Mirror and have to reinstall everything from scratch.

> magicmirror@2.16.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[06.07.2021 14:22.58.440] [ERROR] App threw an error during load
[06.07.2021 14:22.58.451] [ERROR] TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:36:38
    at /home/pi/MagicMirror/js/logger.js:19:20
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:24:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:52:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
A JavaScript error occurred in the main process
Uncaught Exception:
TypeError: Bind must be called on a function
    at bind (<anonymous>)
    at /home/pi/MagicMirror/js/logger.js:36:38
    at /home/pi/MagicMirror/js/logger.js:19:20
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:24:3)
    at Object.<anonymous> (/home/pi/MagicMirror/js/logger.js:52:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)

sdetweil, 2021-07-06:
can you do 

grep version ~/MagicMirror/node_modules/console_stamp/package.json

mine on 2.16 is 3.0.2

what modules are you using? MMM-Logging possibly?

TheBassfly, 2021-07-06:
Mine seems to be 3.0.2 too.

For now I only have standard Module enabled. All third Party modules are removed from the config.json.

sdetweil, 2021-07-06:
also, could you list the last 2 lines of 

~/MagicMirror/js/logger.js
 mine are 
```
	return logLevel;
});

```
lines 49, and 50, with a trailing blank line

the stack trace shows something from line 52
```
at /home/pi/MagicMirror/js/logger.js:36:38
at /home/pi/MagicMirror/js/logger.js:19:20
at Object. (/home/pi/MagicMirror/js/logger.js:24:3)
at Object. (/home/pi/MagicMirror/js/logger.js:52:3)
```

TheBassfly, 2021-07-06:
Last 2 Lines:
	return logLevel;
});

There is nothing below Line 50

sdetweil, 2021-07-06:
ok, one last test.. edit ~/MagicMirror/js/logger.js and comment out line 36 

```
// timeStamp: Function.prototype.bind.call(console.timeStamp, console)
```
I don't know what the side effect will be

what does the config.js logLevel:  setting look like? 
mine in 2.15 looks like 
```
    logLevel: [
      "INFO",
      "LOG",
      "ERROR"
    ],
```

TheBassfly, 2021-07-06:
> ok, one last test.. edit ~/MagicMirror/js/logger.js and comment out line 36
>
> ```
> // timeStamp: Function.prototype.bind.call(console.timeStamp, console)
> ```
>
> I don't know what the side effect will [...]

After the comment out, the logger.js file seems to be empty now.

TheBassfly, 2021-07-06:
Okay, it looks like, we are stuck here. 
Is there any special routine to uninstall MagicMirror oder is it just deleting the MagicMirror Folder in the User Folder?

TheBassfly, 2021-07-06:
When I now run NPM Install I get a bunch of errors:

npm ERR! code ENOSPC
npm ERR! syscall mkdir
npm ERR! path /home/pi/MagicMirror/vendor/node_modules/.staging
npm ERR! errno -28
npm ERR! nospc ENOSPC: no space left on device, mkdir '/home/pi/MagicMirror/vendor/node_modules/.staging'
npm ERR! nospc There appears to be insufficient space on your system to finish.
npm ERR! nospc Clear up some disk space and try again.
npm ERR! code ENOSPC
npm ERR! syscall write
npm ERR! errno -28
npm ERR! nospc ENOSPC: no space left on device, write
npm ERR! nospc There appears to be insufficient space on your system to finish.
npm ERR! nospc Clear up some disk space and try again.
npm ERR! code ENOSPC
npm ERR! syscall write
npm ERR! errno -28
npm ERR! nospc ENOSPC: no space left on device, write
npm ERR! nospc There appears to be insufficient space on your system to finish.
npm ERR! nospc Clear up some disk space and try again.
npm ERR! code ENOSPC
npm ERR! syscall write
npm ERR! errno -28
npm ERR! nospc ENOSPC: no space left on device, write
npm ERR! nospc There appears to be insufficient space on your system to finish.
npm ERR! nospc Clear up some disk space and try again.
npm ERR! code ENOSPC
npm ERR! syscall write
npm ERR! errno -28
npm ERR! nospc ENOSPC: no space left on device, write
npm ERR! nospc There appears to be insufficient space on your system to finish.
npm ERR! nospc Clear up some disk space and try again.
npm ERR! code ELIFECYCLE
npm ERR! errno 228
npm ERR! magicmirror@2.16.0 install: `echo "Installing vendor files ...
npm ERR! " && cd vendor && npm install --loglevel=error`
npm ERR! Exit status 228
npm ERR! 
npm ERR! Failed at the magicmirror@2.16.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

sdetweil, 2021-07-06:
just delete the MM folder..

use my install script after.. it didn't have a problem..

wonder what the upgrade script would have done

see 
https://github.com/sdetweil/MagicMirror_scripts

sdetweil, 2021-07-06:
well, THAT 
npm ERR! nospc ENOSPC: no space left on device, 

is telling.. 

do a df

TheBassfly, 2021-07-06:
What is DF?

sdetweil, 2021-07-06:
how old is the SD card... 

sadly the memory technology is write once.  when it comes time to write the same file again, those memory bits are marked used, but the linux file system doesn't know... 

eventually you run out of bits and things become weird.. 
only fix is a new memory card

sdetweil, 2021-07-06:
>What is DF?

disk full

TheBassfly, 2021-07-06:
> how old is the SD card...
> 
> sadly the memory technology is write once. when it comes time to write the same file again, those memory bits are marked used, but the linux file system doesn't know...
> 
> eventually you run out of bits and things become weird..
> only fix is a new memory card

It is three years now old with 16GB.

sdetweil, 2021-07-06:
i think you have hit the used card wall.. 

I have gone thru 10 memory cards, 8, 16 ad 32 gb  in 4 years now.. 

sd cards are great for camera and music, seldom write, lots of read..

on the two systems where I lost the most cards, I have moved to boot from read only sd card,, 
run from ssd. (which has the same memory problem, but WAY more bits)
but I also have a 5 pack or 32g cards ready to go in case..

TheBassfly, 2021-07-06:
> i think you have hit the used card wall..
> 
> I have gone thru 10 memory cards, 8, 16 and 32 gb in 4 years now..
> 
> sd cards are great for camera and music, seldom write, lots of read..
> 
> on the two systems where I lost the most cards, I have moved to boot from read only sd card,,
> run from ssd. (which has the same memory problem, but WAY more bits)
> but I also have a 5 pack of 32g cards ready to go in case..

I check. I did a cleanup with apt-get. I forgot this after the big upgrade today and after it a got 1.5gb back. I just reinstall MM with your script. Let's see.

TheBassfly, 2021-07-06:
Allright.
It seemed to be the SD Card Issue.
After Cleanup and reinstall of Magic Mirror, end putting back my Backuped files, everything works again.

Thank you all for the patience.

closed 2021-07-06 by TheBassfly

sdetweil, 2021-07-06:
well, make a copy asap.. 

the sd card issue is not solvable.