MagicMirrorOrg/MagicMirror #2610: Problem with the custom.css with the latest version 2.16.0

Titi9562, 2021-07-05:
Hey there,

After updating my Magic Mirror with the 2.16.0 release, my custom.css file does not seem to be taken into account.

The display and organization of my modules is therefore not the same as before.
When I try to modify the custom.css there is no effect on the page ...

Do you have any idea where it can come from?
I saw that the last version made changes on the custom.css.

thanks a lot for your help !

MichMich, 2021-07-05:
Nothing has changed on how the custom css is used. Are you sure it's on the correct location?

Titi9562, 2021-07-05:
I did not move or rename the custom.css file. It's strange, I only realized the update and since then everything is disorganized on my page.

MichMich, 2021-07-05:
It might be worth it to check if it's loaded correctly using the inspector. 

Cmd + Opt + I or Ctrl + Shift + I

closed 2021-10-02 by stale[bot]