MagicMirrorOrg/MagicMirror #2615: Sense of code coverage

khassel, 2021-07-14:
Code coverage is executed with every PR but I don't really know if this is helpful. E.g. why is the coverage so bad (63.63%, before 70.47%) in [this PR](https://github.com/MichMich/MagicMirror/pull/2613)?

We are using `nyc` but with `jest` we have a second build in code cov. So I testet locally the dev branch with both and here are the resuls:

````
nyc: `npm run test:coverage`
------------------------------------|---------|----------|---------|---------|-----------------------------------------------------------------------------------------------------
File                                | % Stmts | % Branch | % Funcs | % Lines | Uncovered Line #s
------------------------------------|---------|----------|---------|---------|-----------------------------------------------------------------------------------------------------
All files                           |   71.22 |     52.7 |   73.48 |    71.2 |
 js                                 |    77.6 |    60.18 |    69.7 |   77.53 |
  app.js                            |   70.59 |       50 |   68.42 |   70.34 | 38-41,73-80,96,130-135,188-199,266-271,279-284
  class.js                          |   63.89 |    68.42 |    62.5 |   63.89 | 13,42-54,94-103
  defaults.js                       |     100 |       75 |     100 |     100 | 82
  deprecated.js                     |     100 |      100 |     100 |     100 |
  electron.js                       |   72.22 |       50 |   41.67 |   72.22 | 40,56,80-90,105-109,116-117,131
  logger.js                         |   91.67 |    66.67 |     100 |   91.67 | 22
  node_helper.js                    |   92.68 |       60 |    87.5 |   92.68 | 22,42,135
  server.js                         |   91.11 |    77.78 |   66.67 |   91.11 | 29-33,72,76
  utils.js                          |     100 |      100 |     100 |     100 |
 modules/default                    |     100 |       50 |     100 |     100 |
  defaultmodules.js                 |     100 |       50 |     100 |     100 | 10
 modules/default/calendar           |   65.25 |    52.31 |   84.38 |   65.34 |
  calendarfetcher.js                |   88.52 |       60 |      80 |   88.52 | 48,54,56,81-83,100
  calendarutils.js                  |   58.49 |    51.98 |   81.82 |   58.56 | ...-384,389,395,403,407,435-436,441,446,452,456,461,465,495-496,515-517,530-531,553,569-574,590-601
  node_helper.js                    |   78.57 |       50 |     100 |   78.57 | 43-45,68-70
 modules/default/newsfeed           |    76.7 |    44.44 |   66.67 |    76.7 |
  newsfeedfetcher.js                |      75 |       36 |   61.11 |      75 | 30,61-66,76-77,93-94,104,116-117,133-134,149
  node_helper.js                    |      80 |    63.64 |   83.33 |      80 | 56-58,65-68
 modules/default/updatenotification |      60 |    31.82 |      80 |   59.18 |
  node_helper.js                    |      60 |    31.82 |      80 |   59.18 | 26-39,58-65,72-75,93,99,114-119
------------------------------------|---------|----------|---------|---------|-----------------------------------------------------------------------------------------------------
````

````
jest: `npx jest -i --forceExit --coverage=true`
---------------------------------|---------|----------|---------|---------|------------------------------------------------
File                             | % Stmts | % Branch | % Funcs | % Lines | Uncovered Line #s
---------------------------------|---------|----------|---------|---------|------------------------------------------------
All files                        |   16.56 |    10.78 |   24.49 |   16.35 |
 js                              |   20.83 |        0 |       0 |   20.83 |
  deprecated.js                  |     100 |      100 |     100 |     100 |
  server.js                      |   15.56 |        0 |       0 |   15.56 | 15-97
  utils.js                       |     100 |      100 |     100 |     100 |
 modules/default                 |     100 |       50 |     100 |     100 |
  defaultmodules.js              |     100 |       50 |     100 |     100 | 10
 modules/default/calendar        |    9.06 |     8.73 |    9.38 |    8.81 |
  calendar.js                    |    9.06 |     8.73 |    9.38 |    8.81 | 69-437,469-728,768,807-844
 modules/default/currentweather  |     1.4 |     1.49 |    5.88 |    1.41 |
  currentweather.js              |     1.4 |     1.49 |    5.88 |    1.41 | 81-582
 modules/default/weatherforecast |    24.3 |    21.92 |      35 |    23.7 |
  weatherforecast.js             |    24.3 |    21.92 |      35 |    23.7 | 74-320,349-352,370-374,387-388,404,432-464,492
 tests/e2e                       |   85.71 |       60 |     100 |   85.71 |
  global-setup.js                |   85.71 |       60 |     100 |   85.71 | 15,23,35,47
 tests/e2e/modules/mocks         |     100 |      100 |     100 |     100 |
  index.js                       |     100 |      100 |     100 |     100 |
  weather_current.js             |     100 |      100 |     100 |     100 |
  weather_forecast.js            |     100 |      100 |     100 |     100 |
 tests/servers                   |     100 |      100 |     100 |     100 |
  basic-auth.js                  |     100 |      100 |     100 |     100 |
 tests/unit/mocks                |     100 |      100 |   23.08 |     100 |
  logger.js                      |     100 |      100 |   23.08 |     100 |
 translations                    |     100 |       50 |     100 |     100 |
  translations.js                |     100 |       50 |     100 |     100 | 53
 vendor                          |     100 |       50 |     100 |     100 |
  vendor.js                      |     100 |       50 |     100 |     100 | 17
---------------------------------|---------|----------|---------|---------|------------------------------------------------
````

So they are totally different.

Jest is using babel and you can switch to another provider v8 (again different results).

So my questions:
- Is the current code cov test really helpful (becoming red in PR's where no one really knows why)?
- can jest be a replacement for nyc?

rejas, 2021-07-14:
Accoring to https://istanbul.js.org/docs/tutorials/jest/ jest has nyc builed into it. So getting rid of the nyc would get a yes from me. 
But why is the percentage so different?

khassel, 2021-07-14:
I see no configuration for nyc in this repo.

Concerning `jest`: It looks like it only reports files which are tested by default.

So I added now

````
                "collectCoverageFrom": [
                        "**/*.{js,jsx}",
                        "!**/node_modules/**",
                        "!**/vendor/**"
                ],
````

to the jest config and here are the results (not better):

````
------------------------------------|---------|----------|---------|---------|------------------------------------------------
File                                | % Stmts | % Branch | % Funcs | % Lines | Uncovered Line #s
------------------------------------|---------|----------|---------|---------|------------------------------------------------
All files                           |     2.5 |     2.43 |    2.05 |    2.71 |
 clientonly                         |       0 |        0 |       0 |       0 |
  index.js                          |       0 |        0 |       0 |       0 | 4-124
 config                             |       0 |        0 |     100 |       0 |
  config.js                         |       0 |        0 |     100 |       0 | 10-111
 coverage/lcov-report               |       0 |        0 |       0 |       0 |
  block-navigation.js               |       0 |        0 |       0 |       0 | 2-79
  prettify.js                       |       0 |        0 |       0 |       0 | 2
  sorter.js                         |       0 |        0 |       0 |       0 | 2-170
 js                                 |    1.12 |        0 |       0 |    1.13 |
  app.js                            |       0 |        0 |       0 |       0 | 9-288
  check_config.js                   |       0 |        0 |       0 |       0 | 8-73
  class.js                          |       0 |        0 |       0 |       0 | 10-108
  defaults.js                       |       0 |        0 |     100 |       0 | 9-83
  deprecated.js                     |     100 |      100 |     100 |     100 |
  electron.js                       |       0 |        0 |       0 |       0 | 3-141
  loader.js                         |       0 |        0 |       0 |       0 | 9-268
  logger.js                         |       0 |        0 |       0 |       0 | 10-49
  main.js                           |       0 |        0 |       0 |       0 | 9-615
  module.js                         |       0 |        0 |       0 |       0 | 10-533
  node_helper.js                    |       0 |        0 |       0 |       0 | 7-146
  server.js                         |   15.56 |        0 |       0 |   15.56 | 15-97
  socketclient.js                   |       0 |        0 |       0 |       0 | 9-51
  translator.js                     |       0 |        0 |       0 |       0 | 9-153
  utils.js                          |     100 |      100 |     100 |     100 |
 modules/default                    |     100 |       50 |     100 |     100 |
  defaultmodules.js                 |     100 |       50 |     100 |     100 | 10
 modules/default/alert              |       0 |        0 |       0 |       0 |
  alert.js                          |       0 |        0 |       0 |       0 | 9-165
  notificationFx.js                 |       0 |        0 |       0 |       0 | 13-157
 modules/default/calendar           |    4.25 |     4.49 |    4.55 |    4.13 |
  calendar.js                       |    9.06 |     8.73 |    9.38 |    8.81 | 69-437,469-728,768,807-844
  calendarfetcher.js                |       0 |        0 |       0 |       0 | 7-158
  calendarutils.js                  |       0 |        0 |       0 |       0 | 11-607
  debug.js                          |       0 |      100 |       0 |       0 | 9-43
  node_helper.js                    |       0 |        0 |       0 |       0 | 7-82
 modules/default/clock              |       0 |        0 |       0 |       0 |
  clock.js                          |       0 |        0 |       0 |       0 | 9-356
 modules/default/compliments        |       0 |        0 |       0 |       0 |
  compliments.js                    |       0 |        0 |       0 |       0 | 7-192
 modules/default/currentweather     |    1.38 |     1.49 |    5.56 |    1.39 |
  currentweather.js                 |     1.4 |     1.49 |    5.88 |    1.41 | 81-582
  node_helper.js                    |       0 |      100 |       0 |       0 | 1-7
 modules/default/helloworld         |       0 |      100 |       0 |       0 |
  helloworld.js                     |       0 |      100 |       0 |       0 | 7-18
 modules/default/newsfeed           |       0 |        0 |       0 |       0 |
  newsfeed.js                       |       0 |        0 |       0 |       0 | 7-391
  newsfeedfetcher.js                |       0 |        0 |       0 |       0 | 7-157
  node_helper.js                    |       0 |        0 |       0 |       0 | 8-83
 modules/default/updatenotification |       0 |        0 |       0 |       0 |
[... 95 lines cut ...]
  index.js                          |     100 |      100 |     100 |     100 |
  weather_current.js                |     100 |      100 |     100 |     100 |
  weather_forecast.js               |     100 |      100 |     100 |     100 |
 tests/servers                      |     100 |      100 |     100 |     100 |
  basic-auth.js                     |     100 |      100 |     100 |     100 |
 tests/unit/classes                 |       0 |      100 |       0 |       0 |
[... cut ...]
  root_path_spec.js                 |       0 |        0 |       0 |       0 | 1-58
 tests/unit/mocks                   |     100 |      100 |   23.08 |     100 |
  logger.js                         |     100 |      100 |   23.08 |     100 |
 translations                       |     100 |       50 |     100 |     100 |
  translations.js                   |     100 |       50 |     100 |     100 | 53
------------------------------------|---------|----------|---------|---------|------------------------------------------------
````

rejas, 2021-07-16:
Mhh... getting coverage for the tests directory seems wrong
Mabye we should only get coverage for
- clientonly
- js 
- modules/default
- serveronly
 directories?

khassel, 2021-07-16:
> we should only get coverage for

no problem to update the config for getting this, but we have still the differences to the current code cov ...

closed 2022-02-28 by khassel