MagicMirrorOrg/MagicMirror #2616: Newsfeed module bug

zoldev, 2021-07-17:
Hello!

I found a bug in newsfeed module.
When i was modify this config: 'truncDescription' and 'lenghtDescription', nothing happen.

Thanks for your answer.

sdetweil, 2021-07-17:
can u show the config entry for newsfeed from config.js

zoldev, 2021-07-17:
`{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "hvg h�rek",
						url: "http://www.hvg.hu/rss/tudomany"
					}
				],
				updateInterval: 60000,
				showSourceTitle: false,
				showPublishDate: false,
				broadcastNewsFeeds: false,
				broadcastNewsUpdates: false,
				showDescription: true,
				wrapTitle: true,
				wrapDescription: true,
				truncDescription: true,
				lenghtDescription: 200
			}
		},`

zoldev, 2021-07-17:
{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "hvg h�rek",
						url: "http://www.hvg.hu/rss/tudomany"
					}
				],
				updateInterval: 60000,
				showSourceTitle: false,
				showPublishDate: false,
				broadcastNewsFeeds: false,
				broadcastNewsUpdates: false,
				showDescription: true,
				wrapTitle: true,
				wrapDescription: true,
				truncDescription: true,
				lenghtDescription: 200
			}
		},

zoldev, 2021-07-17:
> can u show the config entry for newsfeed from config.js

{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "hvg h�rek",
						url: "http://www.hvg.hu/rss/tudomany"
					}
				],
				updateInterval: 60000,
				showSourceTitle: false,
				showPublishDate: false,
				broadcastNewsFeeds: false,
				broadcastNewsUpdates: false,
				showDescription: true,
				wrapTitle: true,
				wrapDescription: true,
				truncDescription: true,
				lenghtDescription: 200
			}
		},

sdetweil, 2021-07-17:
ok, this one is wrong
```
lenghtDescription: 200
```
misspelled,h & t reversed,  should be 
```
lengthDescription:200
```

showDescription:true 
is the default, 

the code in the template is 
```
            {% if config.showDescription %}
                <div class="newsfeed-desc small light{{ ' no-wrap' if not config.wrapDescription }}">
                    {% if config.truncDescription %}
                        {{ description | truncate(config.lengthDescription) }}
                    {% else %}
                        {{ description }}
                    {% endif %}
                </div>
            {% endif %}
```

zoldev, 2021-07-17:
:D ooh my god! :D
THANKS!

closed 2021-07-17 by zoldev

rejas, 2021-07-17:
While you are here @zoldev might I ask you if you could help me out? See, I am wondering why people often dont seem to answer the questions we ask them in the issue template. Those questions are really helpful and can help get an answer faster. So I am wondering if the people dont see them (for technical reasons) or if the language is too complicates or the text is too long? NO offence ment, just hoping to improve that template. Thanks in advance!