MagicMirrorOrg/MagicMirror #2618: forum won't send change password email to user

sdetweil, 2021-07-25:
user trying to login, did thru github, but that not working.

I can email him using his account email and it goes thru, and he replied (both on gmail.com)

user kiina

can someone check the forum logs? I cannot

sdetweil, 2021-07-25:
works for my email

sdetweil, 2021-07-25:
there are no blocks or bans on the user

MichMich, 2021-07-25:
Can he login with user&pass?

Kiina, 2021-07-25:
Well login with user and password won't work since I used github oauth in the past and don't have an password. So I need to reset it to even try to login with it.

MichMich, 2021-07-25:
Is that an option to try? I know there have been some issues with the GitHub connection in the past.

Kiina, 2021-07-25:
Well the github login option is gone, that's why i'm kinda stuck in that weird limbo of no chance to login via github but also no chance to get a password to login via email and password

MichMich, 2021-07-25:
Ah, ok... I get the issue now. Let me check.

MichMich, 2021-07-25:
@Kiina could you give it a try to see if you can now request an password via email. It turns out the email service was broken.

Kiina, 2021-07-25:
Works great, thanks!

MichMich, 2021-07-25:
Good to know! Thanks! Can I close this issue?

Kiina, 2021-07-25:
Yes, thanks!

closed 2021-07-25 by MichMich