MagicMirrorOrg/MagicMirror #2619: Black Screen When Running: Remote Control Module(Jopyth)  https://github.com/Jopyth/MMM-Remote-Control

rfaithe, 2021-07-27:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror installer

If you are facing an issue or found a bug while trying to install MagicMirror via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror Docker image

If you are facing an issue or found a bug while running MagicMirror inside a Docker container please create an issue in the GitHub repository of the MagicMirror Docker image:
[https://github.com/bastilimbach/docker-MagicMirror](https://github.com/bastilimbach/docker-MagicMirror)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).
Rasberry Pi 4

**Node Version**: Make sure it's version 10 or later.

**MagicMirror Version**: Please let us now which version of MagicMirror you are running. It can be found in the `package.log` file.
Magic Mirror 2

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.
1. Dowloaded Module
2. Copied Config code
3. Copied Ip Address from device I wish to control in the ipWhiteList section
4. Saved and Rebooted
5. Hit NPM start and it goes black screeb

**Expected Results**: Describe what you expected to see.
Magic Mirror Display Modules and have access to it using smart phone. Smart Phone is unable to connect to Magic Mirror

**Actual Results**: Describe what you actually saw.
Black Screen, Mirror never displays

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

var config = {
address: '0.0.0.0', // Address to listen on, can be:
// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
// - another specific IPv4/6 to listen on a specific interface
// - "0.0.0.0", "::" to listen on any interface
// Default, when address config is left out or empty, is "localhost"
port: 8080,
basePath: "/", // The URL path where MagicMirror is hosted. If you are using a Reverse proxy
// you must set the sub path here. basePath must end with a /
ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.0.xx", "::ffff:192.168.0.xx"],
// or add a specific IPv4 of 192.168.1.5 :
// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

useHttps: false, // Support HTTPS or not, default "false" will use HTTP
httpsPrivateKey: "", // HTTPS private key path, only require when useHttps is true
httpsCertificate: "", // HTTPS Certificate path, only require when useHttps is true

language: "en",
locale: "en-US",
logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
timeFormat: 24,
units: "metric",
// serverOnly:  true/false/"local" ,
// local for armv6l processors, default
//   starts serveronly and then starts chrome browser
// false, default for all NON-armv6l devices
// true, force serveronly mode, because you want to.. no UI on this device

modules: [
{
module: "alert",
},
{
module: "updatenotification",
position: "top_bar"
},
{
module: "clock",
position: "top_left"
},
{
module: "calendar",
header: "Google Calendar",
position: "top_left",
config: {
calendars: [
{
symbol: "calendar-check",
url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"
},
{
symbol: "MyCalender",
url: "xxx"
}
]
}
},
{
module: "compliments",
position: "lower_third"
},
{
    module: 'MMM-bitcoin',
    position: 'top_right',
    config: {
      fiat: 'usd',           // 'usd' and 'eur' available, defaults to 'usd'
      showBefore: 'Bitcoin',     // will display before the bitcoin price, default 'Bitstamp'
      updateInterval: 60000 // update interval in milliseconds
    }
},
{
  module: "MMM-NowPlayingOnSpotify",
  position: "top_left",

  config: {
    clientID: "xxxxx",
    clientSecret: "xxxxx",
    accessToken: "xxx"
    refreshToken: "xxx"
  }
},
{
module: "weather",
position: "top_right",
config: {
weatherProvider: "openweathermap",
type: "current",
location: "xxx",
//locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
apiKey: "xxxxx"
}
},
{
module: "weather",
position: "top_right",
header: "Weather Forecast",
config: {
weatherProvider: "openweathermap",
type: "forecast",
location: "xxx",
//locationID: "", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
apiKey: "xxxxx"
}
},
{
    module:     'MMM-3Day-Forecast',
    position:   'top_right',
config: {
api_key:    'xxxx',
lat:       xxxx,
lon:        -xxx,
units:      'M',
lang:       'en',
interval:   900000
}
},
{
    module: 'MMM-Remote-Control',
    // uncomment the following line to show the URL of the remote control on the mirror
    // position: 'bottom_left',
    // you can hide this module afterwards from the remote control itself
    config: {
        customCommand: {},  // Optional, See "Using Custom Commands" below
        showModuleApiMenu: true, // Optional, Enable the Module Controls menu
        secureEndpoints: true, // Optional, See API/README.md
        // uncomment any of the lines below if you're gonna use it
        // customMenu: "custom_menu.json", // Optional, See "Custom Menu Items" below
        // apiKey: "", // Optional, See API/README.md for details
        // classes: {} // Optional, See "Custom Classes" below
    }
},
   

]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

MichMich, 2021-07-27:
Please open an issue in the respective repository. This is not a Core issue.

closed 2021-07-27 by MichMich

sdetweil, 2021-07-27:
@rfaithe come to the forums. I'm sure we can help u there

rfaithe, 2021-07-27:
Will do!

Sincerely,

Ryan Faithe

sdetweil, 2021-07-27:
@rfaithe the forums are here
https://forum.magicmirror.builders/