MagicMirrorOrg/MagicMirror #2620: Weather Module - ignoreToday: true Option stops display of the MMM-HTMLBox module

TeNNo2k5, 2021-07-30:
I have spent two hours desperately searching for the error why my local html file is no longer displayed via the "MMM-HTMLBox" module.

**Platform**: Raspberry Pi 3, Rasbian 5.10.52-v7+
**Node Version**: v10.24.1
**MagicMirror Version**: 2.16.0

**Description**: 
As soon as the ignoreToday: true, option is selected, the HTML display disappears (regardless of the content).

**Steps to Reproduce**: 
Attached is a sample code for both modules using the config.js.sample:

`/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/getting-started/configuration.html#general
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
	address: "localhost", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-US",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"					}
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third"
		},
		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				weatherProvider: "openweathermap",
				type: "forecast",
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "YOUR_OPENWEATHER_API_KEY"
			}
		},
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true,
				broadcastNewsFeeds: true,
				broadcastNewsUpdates: true
			}
		},
				{
			//disabled: true,
			module: "weather",
			position: "top_right",
			config: {
				ignoreToday: true,
				type: "daily",
				colored: true,
				location: "",
				locationID: "", //Location ID from http://bulk.openweathermap.org/sample/city.list.json.gz
				apiKey: ""
			}
		},
				{
		module: "MMM-HTMLBox",
		position:"bottom_left",
		config: {
			width: "110%",
			height: "200px",
			refresh_interval_sec: 60,
			content: "",
			file: "sprit.html", //When content is too long, you can use this for external file. relative path from `MMM-HTMLBox`
		}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
`

rejas, 2021-08-01:
Thanks for the report @TeNNo2k5 I think I found the issue, will open a PR soonish

rejas, 2021-08-03:
My fix got merged, so in the develop branch the issue is fixed (if you cant wait for the next release at 01/09/2021)

TeNNo2k5, 2021-08-03:
Thanks Rejas for your super fast code adjustment, since the bug is not system critical for me I'll wait for the next release.

Greetings
TeNNo2k5

rejas, 2021-08-04:
So could you close this ticket @rejas ?

PR #2621 (2021-08-02): Fix undefined error with ignoreToday option in weather module
Move logic for handling splicing forecast data when ignoreToday is set in weathermodule to template.

Otherwise we get an undefined error since the forecast data isnt available on the first getTemplateData call.

Fixes #2620
closed 2021-08-04 by MichMich