MagicMirrorOrg/MagicMirror #2623: Can not start magic mirror with calendar module any longer

jupadin, 2021-08-04:
Dear all,

after a while I tried to start my magic mirror again with the calendar module but out of no reason I currently get the following error:

```
[04.08.2021 12:13.12.814] [LOG]   Starting MagicMirror: v2.16.0
[04.08.2021 12:13.12.821] [LOG]   Loading config ...
[04.08.2021 12:13.12.826] [LOG]   Loading module helpers ...
[04.08.2021 12:13.12.827] [LOG]   No helper found for module: alert.
[04.08.2021 12:13.12.948] [LOG]   Initializing new module helper ...
[04.08.2021 12:13.12.949] [LOG]   Module helper loaded: updatenotification
[04.08.2021 12:13.12.950] [LOG]   No helper found for module: clock.
[04.08.2021 12:13.13.364] [ERROR] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'crypto-js'
[04.08.2021 12:13.13.365] [LOG]   Loading module helpers ...
[04.08.2021 12:13.13.366] [LOG]   No helper found for module: alert.
[04.08.2021 12:13.13.367] [LOG]   Initializing new module helper ...
[04.08.2021 12:13.13.375] [LOG]   Module helper loaded: updatenotification
[04.08.2021 12:13.13.376] [LOG]   No helper found for module: clock.
[04.08.2021 12:13.13.380] [ERROR] Whoops! There was an uncaught exception...
[04.08.2021 12:13.13.385] [ERROR] { Error: Cannot find module 'crypto-js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/digest-fetch/digest-fetch-src.js:10:18)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12) code: 'MODULE_NOT_FOUND' }
[04.08.2021 12:13.13.386] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[04.08.2021 12:13.13.387] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
```

After removing the calendar module, everything works as expected.

Even with the default config.js the error appears.
Any idea how to solve this ?
Even the npm run config:check is ok.

Thanks in advance !

sdetweil, 2021-08-04:
missing crypto-js lib

[04.08.2021 12:13.13.364] [ERROR] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'crypto-js'

do
```
cd ~/MagicMirror
npm install crypto-js
```

jupadin, 2021-08-04:
Okay.
Without the calendar module everything works as expected.
So this made me believe that the calendar module needs it somehow.
But the hint with the dependency of a dependency might be the reason...

MichMich, 2021-08-04:
Are you using the default calendar module?

jupadin, 2021-08-04:
Yes.
I even tried it with the default configuration file provided - but even with this configuration file the error arises.
So the library is somewhere needed and I'm gonna install it as @sdetweil suggested.

sdetweil, 2021-08-04:
@jupadin @MichMich   
its needed by 
```
node_modules/digest-fetch/digest-fetch-src.js
```

MichMich, 2021-08-04:
If I run `npm ls --all` it doesn't show `crypto-js`, so it doesn't seem to be a (deep) dependency of MagicMirror.

MichMich, 2021-08-04:
@sdetweil, that seems to be `crypt` not `crypto-js`, is it?

sdetweil, 2021-08-04:
@MichMich 
the error message is 
```
 Error found: Error: Cannot find module 'crypto-js'
```

sdetweil, 2021-08-04:
maybe the server side changed their encryption type..

MichMich, 2021-08-04:
Yes, I know. But I can't seem to find which module is trying to use it. Since it isn't in the dependency tree.

MichMich, 2021-08-04:
Maybe md5 used `crypto.js` and switched to `crypt`?

MichMich, 2021-08-04:
@jupadin what you could try is reinstalling your node modules to make sure all your dependencies are up to date:

```bash
# from within your ~/MagicMirror folder:

rm -Rf node_modules && npm install
```

sdetweil, 2021-08-04:
calendarfetcher.js uses digest-fetch

MichMich, 2021-08-04:
Yes, which uses `md5`, which uses `crypt`. I can't seem to find any module that has `crypto.js` as a dependency.

MichMich, 2021-08-04:
Sorry. You're correct. But same story! `crypt` is the dependency. not `crypto` (notice the **o** at the end).

sdetweil, 2021-08-04:
md5 mentions crypto-js in the README

MichMich, 2021-08-04:
Yes, but only that it was based on that implementation:
> The original implementation can be found in the CryptoJS project.

It is not a dependency. So I wonder which module is trying to load the crypto-js module.

sdetweil, 2021-08-04:
well, the stack trace says

```
 at Object.<anonymous> (/home/pi/MagicMirror/node_modules/digest-fetch/digest-fetch-src.js:10:18)
```
and turning off calendar stops the error..

MichMich, 2021-08-04:
Yeah. But I'm trying to understand why (especially since @jupadin is the only user with this issue).

jupadin, 2021-08-04:
I just updated the node version as well as the npm version with a clean
```
# from within your ~/MagicMirror folder:

rm -Rf node_modules && npm install
```
which fixes the error...
But anyway, might be interesting where this crypto-js library comes from

MichMich, 2021-08-04:
Found it! It was a dependency of an old `digest-fetch`:

https://github.com/devfans/digest-fetch/pull/24/files

MichMich, 2021-08-04:
So by updating `digest-fetch` the problem went away for you. And since I already had the new version. It didn't appear in my dependency tree.

MichMich, 2021-08-04:
Can I close this issue?

jupadin, 2021-08-04:
Yes. Thank you very much for your time and investigation !!! :)