MagicMirrorOrg/MagicMirror #2624: MagicMirror clock is off in weather module, is correct on the clock itself

TheScottCasey, 2021-08-05:
I'm having troubles configuring MagicMirror.
**Platform**: Raspberry Pi 3
**Node Version**: Make sure it's version 10 or later.

**Description**: The time is correct on the clock in the upper left-hand corner (Eastern Standard Time), but the time as it appears in the weather module in the upper-right is off by varying times, usually about 80 minutes off. If it's possible to remove the clock in the weather module altogether, or to just sync it to the correct time, please let me know. I have not touched any other files except the config file and a custom compliments file.

Here is what my config file looks like. minus the location-specific weather information (which is running correctly otherwise): 

[image]

My Drive
/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * see https://docs.magicmirror.builders/getting-started/configuration.html#general
 * and https://docs.magicmirror.builders/modules/configuration.html
 */
let config = {
	address: "localhost", 	// Address to listen on, can be:
							// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
							// - another specific IPv4/6 to listen on a specific interface
							// - "0.0.0.0", "::" to listen on any interface
							// Default, when address config is left out or empty, is "localhost"
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
					// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], 	// Set [] to allow all IP addresses
															// or add a specific IPv4 of 192.168.1.5 :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
															// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
															// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],

	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "en",
	locale: "en-US",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
	timeFormat: 12,
	units: "imperial",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left"
		},
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
				calendars: [
					{
						symbol: "calendar-check",
						url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"					}
				]
			}
		},
		{
  module: 'compliments',
  position: 'lower_third',
  config: {
                                remoteFile: 'compliments.json'
                        }
},
		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				location: "XXXXX",
				locationID: "XXXXX", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "XXXXX"
			}
		},
		{
			module: "weather",
			position: "top_right",
			header: "Weather Forecast",
			config: {
				weatherProvider: "openweathermap",
				type: "forecast",
				location: "XXXXX",
				locationID: "XXXXX", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "XXXXX"
			}
		},
	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

TheScottCasey, 2021-08-05:
Addendum: I found that I can just remove 

{
module: "weather",
position: "top_right",
config: {
weatherProvider: "openweathermap",
type: "current",
location: "XXXXX",
locationID: "XXXXX", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
apiKey: "XXXXX"
}
},

and have the rest show up fine, but I don't know where the time is pulling from in that string of code. I checked current.js which is only reading as config.Timeformat, which I believe is the same as in clock.js . I can leave it the way it is, but removing that portion of the code entirely leaves out the "feels like" and the current temperature, i.e. everything in the upper-right hand corner of the first screenshot.

sdetweil, 2021-08-05:
that clock is sunrise or sunset time. not current wall clock time

closed 2021-08-31 by MichMich