MagicMirrorOrg/MagicMirror #2629: Calendar: end date missing for multi day events (like holidays) (followup to #2529)

waldmeisterx, 2021-08-09:
Platform: Raspberry Pi 3, current OS
uname -r: 5.10.52-v7+

Node Version: what is this? how do I get this info?

MagicMirror Version: 2.16.0 (problem occured after upgrading from 2.13.0, problem still existed in 2.15.0).

Description: I show holidays with start and end dates. The shown config worked well with 2.13.0, after upgrading to 2.15.0/2.16.0 the end date disappeared. Now, only the current day of the current holiday is shown.

Steps to Reproduce: With my config in MM 2.16.0

Expected Results: start (or current) date AND end date "Sa 10. Apr. - Fr. 16. Apr."

Actual Results: only start (or current) date "Sa 10. Apr."

Configuration:
```
{
	module: "calendar",
	header: "Ferien Hessen",
	position: "top_bar",
	config: {
		tableClass: "medium",
		colored: true,
		fetchInterval: 900000,
		fullDayEventDateFormat: "dd DD. MMM",
		timeFormat: "absolute",
		urgency: 0,
		getRelative: 0,
		maximumEntries: 1,
		maxTitleLength: 50,
		showEnd: true,
		dateEndFormat: "dd DD. MMM",
		calendars: [
			{
				// Ferien Hessen 2021
				symbol: "democrat",
				color: "#ffbbbb",
				url: "https://www.ferienwiki.de/exports/ferien/2021/de/hessen"
			},
		]
	}
},

```
Additional Notes:
The option "showEnd" is shown two times in the Calenders doku with similiar descritions!

sdetweil, 2021-08-10:
@waldmeisterx 
>The option "showEnd" is shown two times in the Calenders doku with similiar descritions!

yes, you can set it once for ALL calendars (above the section calendars:)
or you can set it individually for each calendar... (in the section where u code url)

all overrides individual

waldmeisterx, 2021-08-15:
@sdetweil ok, good to know, that ALL overrides individual, I'd have expected the opposite behaviour.
And for me it is not clear in docu, the entries I see are both in the same table "Configuration options", 4 lines away from each other. 

So you agree, my config should show an end date? It has showEnd = true in ALL and I added it in calender too for this next test

No end date appear:
[image]

It seems, the parameter does only shows end TIME, but not end day of multi day events (like holidays).
[image]
The last line here is a multi day event.

sdetweil, 2021-08-15:
as I understand it, showEnd is for multi-day events only.  showOngoing is for multi-day that has started but not ended. 

I spent a lot of time working on getting the data right. but not the presentation of the data.

waldmeisterx, 2021-08-15:
I added (hidden?) parameter showOngoing = true but no change.
As you describe showOngoing should be the one, the holidays are now and ongoing.

sdetweil, 2021-08-15:
what do you mean by 'hidden'?  its in the doc. and its showOngoing:true, right? 
and its only for events that HAVE STARTED not WILL start

download the ics (put the calendar url in a browser) and edit the ics file (its just text)

show me the BEGIN:VEVENT thru END:VEVENT for the holiday 
it should have a RRULE: that shows repeating on some schedule

sdetweil, 2021-08-15:
BTW the documentation style is module parms 1st (override any following), and calendar parms second, same for RSS feeds ..

waldmeisterx, 2021-08-16:
The docs I'm looking in is this https://docs.magicmirror.builders/modules/calendar.html
There I can not find any info for a parameter showOngoing

In the ics I do not find a RRULE, because it's not repeating. It is our summer holidays lasting 6 weeks and I'd like to see how long they last. 
I'd expect to see the value of DTEND.

```
BEGIN:VEVENT
DTSTAMP:20210816T070023Z
UID:6ec55230-6942-4ce3-9c39-391448b654ba
DTSTART;VALUE=DATE:20210719
DTEND;VALUE=DATE:20210828
DESCRIPTION:Sommerferien 2021 in Hessen - Importiert von Ferienwiki.de
SUMMARY:Sommerferien
END:VEVENT

```

sdetweil, 2021-08-16:
thanks.. very helpful...  I created a matching one with an RRULE.. 
i see they flipped ongoing to now hideOngoing vs showOngoing.. 

but the net is, I think there is a bug , the code (in calendar.js)
```
				if (this.config.showEnd) {     //  if end requested, format it 
						timeWrapper.innerHTML += "-";
						timeWrapper.innerHTML += this.capFirst(moment(event.endDate, "x").format(this.config.dateEndFormat));
					}  /* FIX else */   // uncomment this 
				// For full day events we use the fullDayEventDateFormat
				if (event.fullDayEvent) { // if this is a full day event, replace content above with this -----regardless of other settings ---- bug.. 
						//subtract one second so that fullDayEvents end at 23:59:59, and not at 0:00:00 one the next day
						event.endDate -= oneSecond;
						timeWrapper.innerHTML = this.capFirst(moment(event.startDate, 
"x").format(this.config.fullDayEventDateFormat));
					}
```

if u add an 'else' in between 
and set the settings like this 
```
		{
			module: "calendar",
			header: "US Holidays",
			position: "top_left",
			config: {
						timeFormat:'absolute',
						showEnd: true,
						dateEndFormat:"MMM Do",
						getRelative:0,
						urgency:0,
				calendars: [
					{
						symbol: "calendar-check",
						//url: "webcal://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics"
						url: "http://localhost:8090/modules/default/calendar/sams-test.ics",

					}
				]
			}
		},
```
then u get the start/end		as u expect

waldmeisterx, 2021-08-16:
@sdetweil Yes! You found it. This correction in calendar.js and the config showEnd: true results in exactly in what I was looking for

[image]

Thank you a lot.

rejas, 2021-08-16:
great to hear this got resolved thx a lot @sdetweil will you create a PR for your fix?

jupadin, 2021-08-19:
> @sdetweil Yes! You found it. This correction in calendar.js and the config showEnd: true results in exactly in what I was looking for
> 
> [image]
> 
> Thank you a lot.

@waldmeisterx: If I correctly understood this issue and given your screenshot, the name of the weekday on which the holidays start, is missing isn't it ?

Can you tell me which icon and color code this is ? I really like it !

sdetweil, 2021-08-19:
@jupadin  the cal entry shown does not show THE event start date, but TODAY thru the end.  we don't know WHEN it started in the past.. could be yesterday, or in this case last month.   you could change the fulldateFormat (check the spelling) to include the day I supposed..

jupadin, 2021-08-19:
Ahhh, got it.
@sdetweil: Thank you for the explanation !!

waldmeisterx, 2021-08-19:
@jupadin: You can find my configures symbol and color in the first post, it is
```
symbol: "democrat",
color: "#ffbbbb",

```
There you also see the set configured formats @sdetweil mentioned too. 

```
fullDayEventDateFormat: "dd DD. MMM",
dateEndFormat: "dd DD. MMM",
```
But I think it should be dateEndFormat according to https://docs.magicmirror.builders/modules/calendar.html

sdetweil, 2021-08-19:
the code actually uses 
```
					// Use dateFormat
					timeWrapper.innerHTML = this.capFirst(moment(event.startDate, "x").format(this.config.dateFormat));
// then above code 
                                        if (this.config.showEnd) {     //  if end requested, format it 
						timeWrapper.innerHTML += "-";
						timeWrapper.innerHTML += this.capFirst(moment(event.endDate, "x").format(this.config.dateEndFormat));
					}  /* FIX else */   // uncomment this 
				// For full day events we use the fullDayEventDateFormat
				if (event.fullDayEvent) { // if this is a full day event, replace content above with this -----regardless of other settings ---- bug.. 
						//subtract one second so that fullDayEvents end at 23:59:59, and not at 0:00:00 one the next day
						event.endDate -= oneSecond;
						timeWrapper.innerHTML = this.capFirst(moment(event.startDate, 
"x").format(this.config.fullDayEventDateFormat));
					}
```
dateFormat for the startDate

waldmeisterx, 2021-08-19:
@sdetweil with dateFormat set it now shows both dates in same format
[image]
I'm very happy with this.

jupadin, 2021-08-20:
Just another question, maybe worth another issue or discussion...
Wouldn't it be more convenient to put those configuration options inside each url ?

The following use case:
I would like to have one calendar, containing birthdays, holidays (multiple day events + single day events (both full days)) as well as usual events inside the (single) calendar.
Given the (multiple full day event) holidays, I would like them to be shown as discussed above (Start + End), (single full day event) holidays should only be shown with its start day.
The birthdays should be shown relative, starting 6 days before the actual event and the usual events as usual.

Since the most of the options can only be set globally, such a use case is not possible - or do I (again) understood something wrong ?

waldmeisterx, 2021-08-20:
you can set up more than one calender, each with its own settings

jupadin, 2021-08-20:
Yes, but given the documentation, it is only possible to set the time format (absolute, relative) for all of the urls of the calendar and not for each of the urls...
Or is there any misunderstanding ?

waldmeisterx, 2021-08-20:
No, i think you are right. You should address it in a new issue (feature request).

waldmeisterx, 2022-01-04:
The code correction (in calendar.js) mentiened by @sdetweil was not considered in the last release 2.18.0?
The format of multi day events is now again broken and has to be fixed again manually :-(

sdetweil, 2022-01-04:
hm. I added the else in one of the other fixes, did I get it wrong?

waldmeisterx, 2022-01-04:
@sdetweil 
In 2.18.0 it is not included.
I reassigned your correction from above and this fixes it again

waldmeisterx, 2022-04-18:
fixed in 2.19.0

closed 2022-04-18 by waldmeisterx

TesterAufLebenszeit, 2022-07-14:
Hello together.

I had the same problems as waldmeisterx but did find a solution without changing the code of _calendar.js_.
It seems that the fix is not working.
I left a comment. Maybe this can be a solution.

rejas, 2022-07-27:
> I left a comment. Maybe this can lead to a solution.

Where did you leave a comment?

PR #2748 (2021-12-24): fix calendar across timezone, showEnd/FullDay
**Please make sure that you have followed these 4 rules before submitting your Pull Request:**

> 1. Base your pull requests against the `develop` branch.
>
done
> 2. Include these infos in the description:
>
timezone correction for repeating events (RRULE) with wrong date, right time, correction used offset hours, not day hours)  (#2623)
if showEnd  and FullDate format specified together showEnd result was overlayed (#2629), add else clause

> - Does the pull request solve a **related** issue?
Fixes: #2629 #2632
PR #2829 (2022-03-29): fix fullday/showend conflict #2629
> 1. Base your pull requests against the `develop` branch.
yes
> 2. Include these infos in the description:
>
> - Does the pull request solve a **related** issue?
> - If so, can you reference the issue like this `Fixes #<issue_number>`?
Fixes #2629

> - What does the pull request accomplish? Use a list if needed.
adds else clause
if showend, then skip check for fullday event
> - If it includes major visual changes please add screenshots.
no visual changes
>

> 4. Don't forget to add an entry about your changes to
>    the CHANGELOG.md file.
updated