MagicMirrorOrg/MagicMirror #2632: Calendar-Module: Timezone calculation

jupadin, 2021-08-20:
Hi everybody,

given the following ical calendar:

```
BEGIN:VCALENDAR
VERSION:2.0
PROID:jupadin
BEGIN:VEVENT
DTSTART;VALUE=DATE:20200101
DTEND;VALUE=DATE:20200101
RRULE:FREQ=YEARLY
SUMMARY:MyBirthday
UID:1ad611b1-d1e7-4d95-859b-b60b4f9fc453
END:VEVENT
END:VCALENDAR
```

as local `.ics` file as well as the following config:
```
{
    module: "calendar"
    position: "bottom_left",
    header: "Geburtstage",
    config: {
        colored: true,
        coloredSymbolOnly: true,
        timeFOrmat: "relative",
        dateFormat: "DD. MMM",
        fullDayEventTimeFormat: "DD-MMM-YYYY",
        getRelative: 7,
        urgency 4,
        calenadrs: [
            {
                url: "webcal://localhost:8080/...",
                symbol: "birthday-cake",
                color: "#FFA500",
                repeatingCountTitle: "Geburtstag",
                symbolClass: "birthdaySymbol",
                titleClass: "birthdayTitle bright",
                timeClass: "birthdayTime, regular",
            }
        ]
    }
}
```

Using the Log-Level option `DEBUG` I can see that the calendar module does not recognize that its guessed time zone is similar to the time zone of the calendar and thus, adds an offset of UTC+2 (2 hours), which can be seen here:
[image]

Is this a bug or how do I have to modify my *.ics file in order to display only the day and prevent the module from calculating a (wrong) time zone offset ?

sdetweil, 2021-08-20:
full day events do not have a timezone, but are expressed in local time. internally we use utc
so u could change the full day format to not show hours

sdetweil, 2021-08-20:
also, the system is case sensitive

        timeFOrmat: "relative"

is a typo, right?

jupadin, 2021-08-20:
> also, the system is case sensitive
> 
> ```
>     timeFOrmat: "relative"
> ```
> 
> is a typo, right?
> 
> but u need it set to "absolute" to control the formatting

Yeah, the
```
timeFormat: "relative"
```
is a typo, thanks!

jupadin, 2021-08-20:
> full day events do not have a timezone, but are expressed in local time. internally we use utc
> so u could change the full day format to not show hours
> 
> your tz IS utc +2, correct?

Okay, but using the ```DEBUG``` option, the MM output shows that there are calculations made with the time zone.
My tz is UTC + 2, yes (to be more specific: CEST).

But given the calendar event, shown above
```
DTSTART;VALUE=DATE:20200822
DTEND;VALUE=DATE:202008222
```
where no times are given, how is it possible that my mirror adds the two hours, since the birthday starts at 0:00, when a full day event does not use any time zone information / calculation ?

jupadin, 2021-08-21:
Isn't it like this that given the ical-entry, the calendar module reads the dates as UTC-Time and adds time zone information based on its guess:
https://github.com/MichMich/MagicMirror/blob/26a76f80d63e5c9746b6db712d6901219f8b6b93/modules/default/calendar/calendarutils.js#L25-L33
such that given the following lines:
https://github.com/MichMich/MagicMirror/blob/26a76f80d63e5c9746b6db712d6901219f8b6b93/modules/default/calendar/calendarutils.js#L74-L90
doubles up the time zone adjustment, since the initial times were interpreted as UTC, even though I would like to have them specified in my own (UTC+2) time zone ?

jupadin, 2021-08-21:
After some investigation it seems like the problem is related to the ```node-ical``` module, since this already returns UTC times, even though it should be local (time zone) dates.

There are already open issues related to this on the ```node-ical```module.

In case someone has a hot fix to this issue... otherwise I think we can close this issue.

sdetweil, 2021-08-21:
as I said before we use utc inside. yes node-ical returns utc

I'm the one that moved mm to node-ical, to fix other problems

jupadin, 2021-08-21:
> as I said before we use utc inside. yes node-ical returns utc

Yes, I will further investigate where my error is...

> I'm the one that moved mm to node-ical, to fix other problems

Yep, I saw that you made a lot of contributions to ```node-ical``` while trying to fix my error.

jupadin, 2021-08-21:
> as I said before we use utc inside. yes node-ical returns utc

Even if I try to specify a time in non-utc, ```node-ical``` interprets it as UTC and it seems that there it only handles either the start date or the end date correctly. (see [here](https://github.com/jens-maus/node-ical/issues/147))

jupadin, 2021-08-22:
... found a bug. Gonna provide a pull request as sonn as possible.

PR #2634 (2021-08-29): [Fix] Calendar-Module: Event start time correction based on time zone
> 1) Base your pull requests against the `develop` branch.

Done.

> 2) Include these infos in the description:
>  * Does the pull request solve a **related** issue?
>  * If so, can you reference the issue like this `Fixes #<issue_number>`?
>  * What does the pull request accomplish? Use a list if needed.
>  * If it includes major visual changes please add screenshots.

PR fixes #2632.
The time zone correction was only applied if the date hour of the event is smaller than the actual time zone offset.
In my case the date hour [... cut]
closed 2021-08-29 by jupadin