MagicMirrorOrg/MagicMirror #2638: Newsfeed intervals become chaotic

jalibu, 2021-08-29:
## I found a bug in MagicMirror

**Platform**: Raspberry Pi OS

**Node Version**: Make sure it's version 10 or later.

**MagicMirror Version**: 2.16.0

**Description**: After an unspecified time, the newsfeed module starts with chaotic fading intervals. It looks like multiple, competing intervals are affecting fading at the same time. It is noticeable that there are fading pauses of different lengths.
A browser refresh fixes this problem temporary (as it refreshes client runtime intervals)

https://user-images.githubusercontent.com/25933231/131243809-40b7f42a-968e-4da2-afe1-a5c4d73ba316.mov

**Steps to Reproduce**: This problem is not reproducible "by hand", but comes by itself after an unpredictable runtime (without further human interactions)

**Expected Results**: Constant fading intervals

**Actual Results**: Fading intervals are totally messed up

**Configuration**: 
```
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "Tagesschau.de",
						url: "https://www.tagesschau.de/xml/rss2"
					},
					{
						title: "Heidelberg.de",
						url: "https://www.heidelberg.de/site/Heidelberg_ROOT/pmrss/407657/Lde/pressemitteilungen.html"
					}
				],
				showSourceTitle: true,
				showPublishDate: false,
				broadcastNewsFeeds: false,
				broadcastNewsUpdates: false,
				showDescription: true
			}
		},
```
**Additional Notes**: -

OberfragGER, 2021-08-29:
Here too.
It only happens with 2.16. WIth MM 2.15 no problem.

I also using MM_Pages to flip the page every 1 Minute. No special options in my config:

`   {
      module: 'newsfeed',
      position: 'top_bar',
      header: '',
      config: {
		showDescription: false,
		wrapDescription: false,
        feeds: [
          {
            title: 'Spiegel Online Eilmeldungen',
            url: 'https://www.spiegel.de/schlagzeilen/tops/index.rss'
          },
          {
            title: 'Golem',
            url: 'https://rss.golem.de/rss.php?feed=RSS2.0',
            encoding: 'ISO-8859-1'
          },
          {
            title: 'Heise',
            url: 'https://www.heise.de/newsticker/heise-atom.xml'
          },
          {
            title: 'Spiegel Online komlett',
            url: 'https://www.spiegel.de/schlagzeilen/index.rss'
          },
          {
            title: 'Computerbase',
            url: 'https://www.computerbase.de/rss/news.xml'
          },
          {
            title: 'Tagesschau',
            url: 'http://www.tagesschau.de/xml/rss2'
          }
        ]
      }
    },`

khassel, 2021-08-29:
Already followed the corresponding [Forum-Thread](https://forum.magicmirror.builders/topic/15348/newsfeeder-is-going-way-to-fast-when-showing-news).

Reading through this thread I can summarize:

- Problem exists since v2.16.0 (v2.15.0 is working correctly)
- not related to setup (docker and pm2 was mentioned)
- only occuring on the screen of the pi (not in remote browser using same pi instance)

Questions:

- are there any foreign modules involved (MMM-Pages was mentioned by some users) or is it also occuring without foreign modules installed?
- can someone test the `develop` branch because the electron version change to v13 (current is v11)?
- are there any errors/warnings in the logs (on the pi and/or in the dev console)?

jalibu, 2021-08-29:
@khassel
Since the problem (as in my case) only occurs after several days of runtime, I have only been able to see it on my productive mirror, but not on my development environment.
I could not detect any errors in the Developer Console. However, since I don't keep an eye on the mirror 24/7, it is of course possible that the cause of the problem lies far back.

MichMich, 2021-08-29:
I think the best thing to do is let someone who has this issue test the dev branch without any 3rd party modules. If this solves the issue, add the 3rd party modules to see if that was the issue. If not the issue seems to be solved by using the dev branch.

Sekisback, 2021-09-05:
> 
> 
> I think the best thing to do is let someone who has this issue test the dev branch without any 3rd party modules. If this solves the issue, add the 3rd party modules to see if that was the issue. If not the issue seems to be solved by using the dev branch.

Which file from the developer branch is the one I need to test?
with the exception of the PirModule my Mirror has no 3rd party Modules

sdetweil, 2021-09-05:
@Sekisback   you just do  
```
git checkout dev
``` 
to get to the dev h branch with the latest pre-release code

rejas, 2021-09-05:
@sdetweil shouldnt it be `develop` instead of `dev`?

sdetweil, 2021-09-05:
@rejas doah!, fixed

Sekisback, 2021-09-06:
on the way, develop branch installed, pm2 shows Version 2.17.0.
Cron job disabled. Now i have to wait till... till what??? the behavior occurred after days/weeks. Is there a way to force it.

Sekisback, 2021-09-11:
good news, looks like that you made it, till now it works like expected

OberfragGER, 2021-09-22:
When its going to be the 2,17 release? Think about tonswitch tonthe dev-branch.

sdetweil, 2021-09-22:
October 1, release every 90 days

Sekisback, 2021-09-22:
Bad news.. we were happy too early... not so wild as before, but a few news are shown like expected with the right time. But some of them rush

hangorazvan, 2021-09-28:
Repeating same news several times in a row. v2.17 dev
Has anyone encountered this problem before?

khassel, 2021-10-01:
New release `v2.17.1` is released, so may some people can update and test if this issue is still present using the new version, thanks.

Sekisback, 2021-10-03:
updated to 2.17.1, now we will see whats happen in the next few days

OberfragGER, 2021-10-04:
After a little time, every 2nd news is flashing. But the others seems to be stablem.
So its working by 50% now ;)

radokristof, 2021-10-07:
I have the same problem. Latest version installed, though the interval is chaotic, article disappears after 3-5sec, but updateInterval is set to 1min. Also some news are also flashing for me.

hangorazvan, 2021-10-10:
I went back to version 2.14 for newsfeed, before nunjucks template, it is more stable and works properly

MichMich, 2021-10-10:
What would be interesting is to test 2.14 but with the latest Electron version. Just to make sure it's not an issue of Electron.

hangorazvan, 2021-10-10:
I don't use electron, I go into server mode and display in chrome Kiosk mode

Sekisback, 2021-10-10:
Using electron, runs without issues since the last update (7days)

jalibu, 2021-10-10:
I am using Electron with 2.17 and still had the problem.
Let me mention one thing: I added a new external WiFi Adapter to my system to improve the internet stability, as I had some connectivity issues (like timeouts etc). Since I did that, the problem did not occur again.

Perhaps the cause of the problem is near the error handling/retry mechanism?

MichMich, 2021-10-10:
What's the update interval you are using?

hangorazvan, 2021-10-10:
> Hmmm that's interesting. How about everyone else with this issue? Using electron or server only?
> 
> Also: why chrome kiosk mode and not electron?

I am running a modified version on windows, on several devices and it seems simpler to me

MichMich, 2021-10-10:
@hangorazvan makes sense.

jalibu, 2021-10-10:
I thought about this change: https://github.com/MichMich/MagicMirror/commit/37fab7ac6374c9468118d31f0fdc9851acc9f111#diff-ff5384d40bfcc5aeb84bf47c5652e5569f09ab0c2aa35486de39712451c5c396

It recalls scheduleUpdateInterval

Sekisback, 2021-10-14:
after 11 days without trouble, same behavior as before.  with the difference, that the news stay a little bit longer may 3second

radokristof, 2021-10-17:
Installed dev version, same as before. After some time, the updateInterval is not normal.

jalibu, 2021-10-17:
I found a way to reproduce the problem:

1. Set the reloadInterval short (to simplify testing, eg. 30 * 1000)
2. Start MM -> Works fine
3. Unplug network connection -> Module will display error when it tries to reload newsfeeds
4. Connect to the Internet again
5. Module will fetch newsfeeds again but with chaotic fading intervals

jalibu, 2021-10-17:
You could also just use the Browser console and execute once:
```
MM.getModules()[0].socketNotificationReceived("NEWSFEED_ERROR", {error_type: 'MODULE_ERROR_UNSPECIFIED'})
````
where getModules()[0] is the id of the newsfeed module

jalibu, 2021-10-17:
Update: Or even simpler,

just execute `MM.getModules()[0].scheduleUpdateInterval()` like the module does, after an error has occurred

sdetweil, 2021-10-17:
> Thanks for the hints @jalibu I might have found the issue behind it. Please check out my PR..

i don't see any open PR's

rejas, 2021-10-17:
> Like I said 7 days ago ;-)

Didnt have the time to look at the whole thread these days :-)

MichMich, 2021-10-17:
🎉

jalibu, 2021-10-17:
Haha, I opened a PR, too. But you @rejas were first.
It seems I will never make it into the desired MagicMirror Contributors Hall of Fame ¯\_(ツ)_/¯ :-D

MichMich, 2021-10-17:
Haha. One day ....

closed 2022-03-01 by jalibu