MagicMirrorOrg/MagicMirror #2639: Calendar-Module: Discussion about recurring title

jupadin, 2021-08-29:
Hi there,

I would like to have a birthday calendar, also showing the age of the person.
Sometimes I don't know their birth year such that I would like to have a possibility to show the age of, whenever I know the birth year, given in the ICS-File, and no recurring title, whenever I don't know the year.

Unfortunately, this is currently not possible, since I have to specify a year.
So I thought I would use a specific year, such that whenever this year is used (of course it has to be before 1900 or sth.), the calendar does not show the age of the person.

My idea was to adapt the way, my calendar is handling this and chose the year `1604`.
However, in lines 252 up to 255 in `calendarutils.js`
https://github.com/MichMich/MagicMirror/blob/26a76f80d63e5c9746b6db712d6901219f8b6b93/modules/default/calendar/calendarutils.js#L251-L255
the date of the event get set to `1990`, since there might happen strange things with events / birthdays before `1900`.

Anyways, even with this "fix", there still happen strange things as shown below:

Given the following ics. file:

```
BEGIN:VCALENDAR
VERSION:2.0
PROID:jupadin
BEGIN:VEVENT
DTSTART;VALUE=DATE:19900303
DTEND;VALUE=DATE:19900304
RRULE:FREQ=YEARLY;INTERVAL=1
SUMMARY:Event with title
UID:d4652c30-769f-4ea9-a7de-f136a44af58b
END:VEVENT
BEGIN:VEVENT
DTSTART;VALUE=DATE:16040101
DTEND;VALUE=DATE:16040102
RRULE:FREQ=YEARLY;INTERVAL=1
SUMMARY:Event without
UID:e1294825-c79e-43dc-afbf-85bbc5857842
END:VEVENT
END:VCALENDAR
```
the parsed string of lines 71 up to 73 in `calendarfetcher.js` is the following:
```
{
  "type": "VEVENT",
  "params": [],
  "start": "1603-12-31T23:59:32.000Z",
  "datetype": "date",
  "end": "1604-01-01T23:06:32.000Z",
  "rrule": {
    "_cache": {
      "all": false,
      "before": [],
      "after": [],
      "between": []
    },
    "origOptions": {
      "dtstart": "1603-12-31T23:59:32.000Z",
      "freq": 0,
      "interval": 1
    },
    "options": {
      "freq": 0,
      "dtstart": "1603-12-31T23:59:32.000Z",
      "interval": 1,
      "wkst": 0,
      "count": null,
      "until": null,
      "bysetpos": null,
      "bymonth": [
        12
      ],
      "bymonthday": [
        31
      ],
      "bynmonthday": [],
      "byyearday": null,
      "byweekno": null,
      "byweekday": null,
      "bynweekday": null,
      "byhour": [
        23
      ],
      "byminute": [
        59
      ],
      "bysecond": [
        32
      ],
      "byeaster": null
    }
  },
  "summary": "Event without",
  "uid": "e1294825-c79e-43dc-afbf-85bbc5857842"
}
```
 
such that the parsed start date of the event / birthday is wrong (which might be related to `node-ical.js` @sdetweil ?).

Any idea how to solve this ?
[image]

sdetweil, 2021-08-29:
everything is based on moment  which is milliseconds since 
 the Unix Epoch (Jan 1 1970 12AM UTC).

MichMich, 2021-08-29:
In your PR you use the arbitrary value 1900. I'm not a big fan of magic values, so maybe there is a better way to solve this. Why value does it return when a birth date is not set?

jupadin, 2021-08-29:
@sdetweil: Then there must be a bug somewhere, since in the `rrule`-field, the year does not get set to `1900`, as supposed here:
https://github.com/MichMich/MagicMirror/blob/26a76f80d63e5c9746b6db712d6901219f8b6b93/modules/default/calendar/calendarutils.js#L251-L255

@MichMich: Me neither. According to the icalendar specification
https://datatracker.ietf.org/doc/html/rfc5545#section-3.3.4
the date is specified as 8 digit value.

Using a mal formatted ics-file, the calendar module reports an error fetching this specific calendar.

Using a year value of `0000`, also returns weird values parsing the ics-file (maybe more `node-ical` related).
Maybe there has to be some preprocessing but since the icalendar format request the date to be a 8 digit value, there has to be some "magic number".

sdetweil, 2021-08-29:
1900 is BEFORE 1970... so there can be NO event starting before Jan 1 1970

jupadin, 2021-08-29:
You are right, 1900 is before 1970 but I'm still able to correctly parse and display an event starting BEFORE 1970.
See below:
```
BEGIN:VCALENDAR
VERSION:2.0
PROID:jupadin
BEGIN:VEVENT
DTSTART;VALUE=DATE:19301111
DTEND;VALUE=DATE:19301112
RRULE:FREQ=YEARLY;INTERVAL=1
SUMMARY:Max Mustermann
UID:5f42ec61-2fae-4ef2-87b8-1852f1545af1
END:VEVENT
END:VCALENDAR
```
and
[image]

However:
If the year is before 1900, the module `node-ical` returns weird values:
```
BEGIN:VCALENDAR
VERSION:2.0
PROID:jupadin
BEGIN:VEVENT
DTSTART;VALUE=DATE:18301111
DTEND;VALUE=DATE:18301112
RRULE:FREQ=YEARLY;INTERVAL=1
SUMMARY:Max Mustermann
UID:5f42ec61-2fae-4ef2-87b8-1852f1545af1
END:VEVENT
END:VCALENDAR
```
gives
```
{
  "type": "VEVENT",
  "params": [],
  "start": "1830-11-10T23:59:32.000Z",
  "datetype": "date",
  "end": "1830-11-11T23:06:32.000Z",
  "rrule": {
    "_cache": {
      "all": false,
      "before": [],
      "after": [],
      "between": []
    },
    "origOptions": {
      "dtstart": "1830-11-10T23:59:32.000Z",
      "freq": 0,
      "interval": 1
    },
    "options": {
      "freq": 0,
      "dtstart": "1830-11-10T23:59:32.000Z",
      "interval": 1,
      "wkst": 0,
      "count": null,
      "until": null,
      "bysetpos": null,
      "bymonth": [
        11
      ],
      "bymonthday": [
        10
      ],
      "bynmonthday": [],
      "byyearday": null,
      "byweekno": null,
      "byweekday": null,
      "bynweekday": null,
      "byhour": [
        23
      ],
      "byminute": [
        59
      ],
      "bysecond": [
        32
      ],
      "byeaster": null
    }
  },
  "summary": "Max Mustermann",
  "uid": "5f42ec61-2fae-4ef2-87b8-1852f1545af1"
}
```
where you can see
```
"start": "1830-11-10T23:59:32.000Z"
```
and
```
"end": "1830-11-11T23:06:32.000Z"
```

Of course there is the thing with
```
"byhour": [
    23
],
"byminute": [
    59
],
"bysecond": [
    32
],
```
which is related to the given date, but I have no clue how :-(.

Given an ics file with year `0000`:
```
BEGIN:VCALENDAR
VERSION:2.0
PROID:jupadin
BEGIN:VEVENT
DTSTART;VALUE=DATE:00001111
DTEND;VALUE=DATE:00001112
RRULE:FREQ=YEARLY;INTERVAL=1
SUMMARY:Max Mustermann
UID:5f42ec61-2fae-4ef2-87b8-1852f1545af1
END:VEVENT
END:VCALENDAR
```

The MagicMirror (re)sets the year to `1900` and correctly displays the data:
[image]

And since the ical format requires and 8 digit date, this is my current solution to the problem.

I also don't like magic numbers but since a year before `1900`, should (!) automatically be set to `1900` there is already some magic happening.

Given the PR, the mirror does not show the age of the person:
[image]

For me, this issue and the related PR can be closed but this would be a real nice feature for further implementations :)

sdetweil, 2021-08-29:
weird results, as expected... 

not MM, but I have a small csv based bd cal module, some have changed the styles to make it look like cal
https://github.com/sdetweil/birthdaylist

shows the age

jupadin, 2021-08-30:
> In your PR you use the arbitrary value 1900. I'm not a big fan of magic values, so maybe there is a better way to solve this. Why value does it return when a birth date is not set?

@MichMich what about having a special tag in the ics file, specifying whether the recurring title should be shown or not.

For example, if the `description` tag contains `showRecurringTitle`, the mirror shows the recurring title ?

MichMich, 2021-08-31:
I think it's a bit too obfuscated for users to understand what they need to do for this feature. In my opinion, there should be a flag in the config of a specific calendar to enable or disable this feature. But then there still is the issue of entries without the correct birthdate. What happens if you use `0000` as year of birth?

jupadin, 2021-08-31:
Yep, a flag in the config would be the perfect solution but since - as you also already said - this feature is entry-related and has to be adapted for each entry.
Using `0000` as year of birth will be reset to `1900` as year of birth due to the lines `251 - 255` in `calendarutils.js`.
Given the ics-file
```
BEGIN:VCALENDAR
VERSION:2.0
PROID:jupadin
BEGIN:VEVENT
DTSTART;VALUE=DATE:00001111
DTEND;VALUE=DATE:00001112
RRULE:FREQ=YEARLY;INTERVAL=1
SUMMARY:Max Mustermann
UID:5f42ec61-2fae-4ef2-87b8-1852f1545af1
END:VEVENT
END:VCALENDAR
```
the MM (using `node-ical`) returns
```
{
  "type": "VEVENT",
  "params": [],
  "start": "1900-11-11T00:00:00.000Z",
  "datetype": "date",
  "end": "1900-11-11T23:00:00.000Z",
  "rrule": {
    "_cache": {
      "all": false,
      "before": [],
      "after": [],
      "between": []
    },
    "origOptions": {
      "dtstart": "1900-11-11T00:00:00.000Z",
      "freq": 0,
      "interval": 1
    },
    "options": {
      "freq": 0,
      "dtstart": "1900-11-11T00:00:00.000Z",
      "interval": 1,
      "wkst": 0,
      "count": null,
      "until": null,
      "bysetpos": null,
      "bymonth": [
        11
      ],
      "bymonthday": [
        11
      ],
      "bynmonthday": [],
      "byyearday": null,
      "byweekno": null,
      "byweekday": null,
      "bynweekday": null,
      "byhour": [
        0
      ],
      "byminute": [
        0
      ],
      "bysecond": [
        0
      ],
      "byeaster": null
    }
  },
  "summary": "Max Mustermann",
  "uid": "5f42ec61-2fae-4ef2-87b8-1852f1545af1"
}
```
such that the year of birth is set to `1900` in both, `start` end `end`.
That was my first try in the PR to implement this feature.

Do you think that such a feature is beneficial and other user would be happy about it ?
It's quite a good use case for birthday calendars but I have the feeling that our possibilities rely on the ics-format.
Following the approach of the apple calendar, given a birthday without a year of birth, it automatically sets `1604` for this specify event (the magic number approach).
Due to using the `node-ical` module, we have to specify a well formatted date according to RFC 5545 (the ical-format).

MichMich, 2021-08-31:
This biggest concern I have is maintenance and future issues. The calendar module is this most error prone module. Adding a feature like this makes it even more error prone, while there probably aren't a lot of users that will use this feature. 

My gut says that it might be better to built a separate (3rd party) birthday module that can handle a calendar feed, and does exactly what you want (and more than what the current calendar can do).

The birthday module could ba based of the current calendar module, although it might be smarter to start with a clean slate.

MichMich, 2021-08-31:
Btw, it would be better to call it an Anniversaries module, because there are other events you might want to add to this module (weddings, etc).

jupadin, 2021-08-31:
Good point with the name, thank you!

Okay, then I will write a 3td party module, based on the calendar module.

closed 2021-08-31 by jupadin

MichMich, 2021-08-31:
👍🏻 Keep me posted on the result. It seems like an interesting module for my own mirror. Also: Facebook used to have a .ical feed for your 'friends' birthdays.