MagicMirrorOrg/MagicMirror #2644: Update notifications not working as intended

Richard238, 2021-08-30:
From - https://forum.magicmirror.builders/topic/15425

`git remote -v`

origin https://github.com/MichMich/MagicMirror.git (fetch)
origin https://github.com/MichMich/MagicMirror.git (push)

config is set to get update notifications:

`{
 module: "updatenotification", position: "top_bar" 
}, `

I received no notification of the upgrade to 2.15, or to 2.16.

When I manually applied an update, I was some 4130 commits behind.

rejas, 2021-08-30:
A quick check seems to indicate that the updatenotification worked until 2.15.0 
With that release something broke so that only module updates are shown but not the core updates.

rejas, 2021-08-30:
Maybe @buxxi can take a look at it since he worked on that module during the developemnt cycle.

khassel, 2021-08-30:
This is caused by https://github.com/MichMich/MagicMirror/pull/2100.

Running with `--dry-run` the property `fetchData.behind` is always = 0.

Running without `--dry-run` works as expected.

So if there is no better idea we have to revert the above PR.

rejas, 2021-08-31:
So @bee-mar should weigh in on this topic :-)

buxxi, 2021-09-02:
I didn't receive a notification for the last release, have been thinking about trying to debug the cause but @khassel seems to have been quicker.

I have one module that hasn't been updated that I tested an alternative way to find out the amount of commits:
```sh
git fetch --dry-run
```
> From https://github.com/timdows/MMM-JsonTable
>   8059e8d..3051433  master     -> origin/master
```sh
git rev-list --ancestry-path 8059e8d..3051433 | wc -l
```
> 4

haven't checked how this would work in the module, but it doesn't actually update the local git with the remote. Wouldn't want every module to auto update when a commit is pushed to the repo :)

khassel, 2021-10-01:
This can be closed, fixed in new release `v2.17.1`.

closed 2021-10-01 by MichMich