MagicMirrorOrg/MagicMirror #2647: 127.0.0.1:8080 already in use

Binog, 2021-09-01:
Hello and thank you for opening an issue.

**Please make sure that you have read the following lines before submitting your Issue:**

## I'm on:

**Platform**:  Linux
**Node Version**: recently updated to 15

**MagicMirror Version**: 16

**Description**: After updating Magicmirror, I do get a black screen with a broken icon.
This is my error-log:

> > 
> > magicmirror@2.16.0 start /home/pi/MagicMirror
> > DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js
> 
> [01.09.2021 18:03.30.489] [LOG]   Starting MagicMirror: v2.16.0
> [01.09.2021 18:03.30.493] [LOG]   Loading config ...
> [01.09.2021 18:03.30.500] [LOG]   Loading module helpers ...
> [01.09.2021 18:03.30.502] [LOG]   No helper found for module: alert.
> [01.09.2021 18:03.30.661] [LOG]   Initializing new module helper ...
> [01.09.2021 18:03.30.661] [LOG]   Module helper loaded: updatenotification
> [01.09.2021 18:03.30.662] [LOG]   No helper found for module: clock.
> [01.09.2021 18:03.30.802] [LOG]   Initializing new module helper ...
> [01.09.2021 18:03.30.803] [LOG]   Module helper loaded: calendar
> [01.09.2021 18:03.30.804] [LOG]   No helper found for module: compliments.
> [01.09.2021 18:03.30.805] [LOG]   No helper found for module: weather.
> [01.09.2021 18:03.30.828] [LOG]   Initializing new module helper ...
> [01.09.2021 18:03.30.828] [LOG]   Module helper loaded: newsfeed
> [01.09.2021 18:03.30.829] [LOG]   All module helpers loaded.
> [01.09.2021 18:03.30.919] [LOG]   Starting server on port 8080 ...
> [01.09.2021 18:03.30.933] [LOG]   Server started ...
> [01.09.2021 18:03.30.934] [LOG]   Connecting socket for: updatenotification
> [01.09.2021 18:03.30.935] [LOG]   Connecting socket for: calendar
> [01.09.2021 18:03.30.935] [LOG]   Starting node helper for: calendar
> [01.09.2021 18:03.30.936] [LOG]   Connecting socket for: newsfeed
> [01.09.2021 18:03.30.937] [LOG]   Starting node helper for: newsfeed
> [01.09.2021 18:03.30.937] [LOG]   Sockets connected & modules started ...
> [01.09.2021 18:03.31.177] [ERROR] Whoops! There was an uncaught exception...
> [01.09.2021 18:03.31.184] [ERROR] Error: listen EADDRINUSE: address already in use 127.0.0.1:8080
>     at Server.setupListenHandle [as _listen2] (net.js:1313:16)
>     at listenInCluster (net.js:1361:12)
>     at GetAddrInfoReqWrap.doListen [as callback] (net.js:1498:7)
>     at GetAddrInfoReqWrap.onlookup [as oncomplete] (dns.js:68:8) {
>   code: 'EADDRINUSE',
>   errno: 'EADDRINUSE',
>   syscall: 'listen',
>   address: '127.0.0.1',
>   port: 8080
> }
> [01.09.2021 18:03.31.185] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
> [01.09.2021 18:03.31.186] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
> [01.09.2021 18:03.31.205] [LOG]   Launching application.
> /home/pi/MagicMirror/node_modules/electron/dist/electron exited with signal SIGINT

**Steps to Reproduce**: I did original (unchanged) config.js.sample as a test for my config.js. Same error !

sdetweil, 2021-09-01:
this means u have MagicMirror already running when U tried to start it again

if u are using pm2 to manage MM running, you need to stop the prior version

pm2 status 
will show u what is running
pm2 stop all 
will stop all tasks

then u can npm start manually or use pm2 start x
where x is the name or number shown in pm2 status

MichMich, 2021-09-01:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2021-09-01 by MichMich

Binog, 2021-09-03:
@sdetweil 
Thank You for Your reply, but not there is just one instance in pm2.

sdetweil, 2021-09-03:
then something else is running on loprt 8080

MichMich, 2021-09-03:
As a quick solution, try restarting your api.